ct-Open-Source/tuya-convert #671: Device did not appear with the intermediate firmware - Moes WS-EU1-LB 1 Gang No Neutral Switch

jimmi1982, 2020-06-15:
Hi,
I purchased one of these Moes 1 Gang No Neutral switch which i checked and was compatible with tuya-convert. I've looked inside and it is running an ESP8285 chip however I have been unable to flash the firmware using tuya-convert.

I'm a relative newby so have been following the various videos on youtube closely. I've tried using both iphone & android tablet for the flashing as I read that sometimes there are issues when using a iOS device.

Below are logs:

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4782376/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4782377/smarthack-udp.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4782378/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4782379/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4782380/smarthack-psk.log)

Any help would be gratefully received.

Thanks

Moisconfused, 2020-06-19:
got a Yagusmart No-neutral 3, 2, and 1 gang, it looks exactly the same as the Moe's line (im assuming its the same just marketed under a different name). I was able to flash all 4 of my 3 gang sand configure them using the Moes no-neutral 3-gang template and it worked perfectly. 

Problem is, for some reason im only able to do that with the 3 gangs, for the life of me i couldnt manage to flash the 1 gang and 2 gang, i get the same error as @jimmi1982 

note: i am an utter noob at this,  not even sure if i should be posting this here, its a long shot but im hoping its the same issue

any help or direction would be amazing, im clearly way out of my league and no idea where to start researching. 

thank you in advance!

kueblc, 2020-06-19:
Duplicate of #483

closed 2020-06-19 by kueblc