ct-Open-Source/tuya-convert #674: Flashing of three Avatar E14 bulbs not possible

carsten-h, 2020-06-18:
### Before opening a new issue, please read the following

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**
Hello!

I have three Avatar E14 bulbs (ALS15L) that I can not flash. 
The link to them is this: [https://www.amazon.de/gp/product/B07W6Z6KPG](url) and here is the entry for tasmota: [https://templates.blakadder.com/avatar_ALS15L.html](url).

I flashed three of them (one was in a package with one of the new ones) before with no problems, but it is a while ago.
With the Smartlife app I can see the following versions:
Hauptmodul: 1.0.2
MCU-Modul: 1.0.2

This is what is happening on the shell:
```
pi@raspberrypi:~/tuya-convert $ sudo ./start_flash.sh 
tuya-convert v2.4.3
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
sudo: ufw: Befehl nicht gefunden
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
....Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
........................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..............................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
```

The bulb was flashion fast at starting and just before the first "SmartConfig complete" is shown it stops flashing and is on all the time.

Here are the logfiles in one file:
[logfiles.txt](https://github.com/ct-Open-Source/tuya-convert/files/4798734/logfiles.txt)

Is there anyone out there who can help me with this?
The strange thing is that it worked three weeks ago with bulbs and switches without any problem.

nightshade00013, 2020-06-18:
I just flashed a set of the Avatar bulbs https://smile.amazon.com/gp/product/B07SBPBDZW and had some issues at first.  What I found is that I had to delete the files for Tuya Convert each time and had to make sure I connected my cell phone to the vtrust-flash network that my computer created before putting the bulb into pairing mode by turning it on and off at least three times with about a second in between.

I however used ESPHome as my firmware provider and had to compile those based on https://esphome.io/cookbook/teckin_sb50.html setup.

elbueno222, 2020-08-12:
> I just flashed a set of the Avatar bulbs https://smile.amazon.com/gp/product/B07SBPBDZW and had some issues at first. What I found is that I had to delete the files for Tuya Convert each time and [...]

What do you mean with "I had to delete the files for Tuya Convert each time", what files did you have to delete?

starwarskidfan, 2020-08-18:
I am having the same issue with different bulbs by the same company.

nightshade00013, 2020-08-19:
> What do you mean with "I had to delete the files for Tuya Convert each time", what files did you have to delete?

Literally removed all of the files, deleted the directory and extracted them again from the archive.  It worked once and then not again.  The other day I got some bulbs from the local big box store that I figured were also Tuya based and they flashed without any issue whatsoever.

adicrescenzo, 2020-10-25:
> Hello!
> 
> I have three Avatar E14 bulbs (ALS15L) that I can not flash.
> The link to them is this: [https://www.amazon.de/gp/product/B07W6Z6KPG](url) and here is the entry for tasmota: [https://templates.blakadder.com/avatar_ALS15L.html](url).
> 
> I flashed three of them (one was in a package with one of the new ones) before with no problems, but it is a while ago.
> With the Smartlife app I can see the following versions:
> Hauptmodul: 1.0.1
> MCU-Modul: 1.0.1
> 
> This is what is happening on the shell:
> 
> ```
> pi@raspberrypi:~/tuya-convert $ sudo ./start_flash.sh 
> tuya-convert v2.4.3
> Checking for network interface wlan0... Found.
> Checking UDP port 53... Available.
> Checking UDP port 67... Available.
> Checking TCP port 80... Available.
> Checking TCP port 443... Available.
> Checking UDP port 6666... Available.
> Checking UDP port 6667... Available.
> Checking TCP port 1883... Available.
> Checking TCP port 8886... Available.
> sudo: ufw: Befehl nicht gefunden
> ======================================================
>   Starting AP in a screen
>   Starting web server in a screen
>   Starting Mosquitto in a screen
>   Starting PSK frontend in a screen
>   Starting Tuya Discovery in a screen
> 
> ======================================================
> 
> IMPORTANT
> 1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
>    This step is IMPORTANT otherwise the smartconfig may not work!
> 2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
>    Make sure nothing else is plugged into your IoT device while attempting to flash.
> 3. Press ENTER to continue
> 
> 
> ======================================================
> Starting smart config pairing procedure
> Waiting for the device to install the intermediate firmware
> ....Put device in EZ config mode (blinking fast)
> Sending SSID                  vtrust-flash
> Sending wifiPassword          
> Sending token                 00000000
> Sending secret                0101
> ........................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ..................
> SmartConfig complete.
> Resending SmartConfig Packets
> ..............................................................................
> Device did not appear with the intermediate firmware
> Check the *.log files in the scripts folder
> Do you want to try flashing another device? [y/N] n
> ======================================================
> Cleaning up...
> Closing AP
> Exiting...
> ```
> 
> The bulb was flashion fast at starting and just before the first "SmartConfig complete" is shown it stops flashing and is on all the time.
> 
> Here are the logfiles in one file:
> [logfiles.txt](https://github.com/ct-Open-Source/tuya-convert/files/4798734/logfiles.txt)
> 
> Is there anyone out there who can help me with this?
> The strange thing is that it worked three weeks ago with bulbs and switches without any problem.

I'm having the same problem with the same light bulbs and a raspberry pi4.
I also noticed that my phone took very long time to join the vtrust-flash wifi network. Only after it happens I put the light bulb in flashing mode but it stops to blink before the first "SmartConfig complete".

Here is the log files, can anyone help?
Thanks in advance

[logfiles.txt](https://github.com/ct-Open-Source/tuya-convert/files/5434286/logfiles.txt)

kevin2xk, 2020-12-08:
> I'm having the same problem with the same light bulbs and a raspberry pi4.
> I also noticed that my phone took very long time to join the vtrust-flash wifi network. Only after it happens I put the [...]

I have the same issue with the E14 bulbs ALS15L. If I'm not mistaken, this is our issue: https://github.com/ct-Open-Source/tuya-convert/issues/483
Seeing the same issues in the logfiles as the above.