ct-Open-Source/tuya-convert #675: Feit "Smart Wifi Bulb" fails to flash

cgull, 2020-06-19:
I can't get this bulb to upgrade, any ideas?

I have a Feit "Smart WiFi Bulb", item number "OM60/RGBW/CA/AG (P)".  This is not from Costco, rather it's a single bulb in a box, from my utility's energy efficiency program.  I don't know what the firmware version is, but these are new, so probably recent.

I'm working with a Raspberry Pi Zero W with the latest Buster image (2020-05-27).  I've now tried both the lite and the full images.  I configure it with USB ethernet emulation and connect it to another Linux system.  I ssh into the RPi for console access.

This lightbulb has never been used with the vendor's software, and has never been allowed to connect to a network with Internet access.

When I run `start_install.sh`, I power on the lightbulb, and it comes up in fast-flashing mode.  I connect an Android phone to the `vtrust-flash` SSID.  I press Enter to start programming the bulb, and I get this:

```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
..Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
...............
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
```

The lightbulb stops flashing shortly after the programming step starts.  The script tries SmartConfig 10 times and gives up.  Power-cycling the lightbulb returns it to fast-flashing mode.  I've gotten it into slow-flashing mode once and tried to program it then, the lightbulb just kept flashing.

Logs are attached.  I also have a tcpdump capture of the tuya-convert AP session if anyone wants it.  The phone's MAC address is `a4:ca:a0:eb:0f:63` so `68:57:2d:9a:1f:9c` is probably the lightbulb.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4803290/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4803291/smarthack-psk.log)

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4803297/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4803302/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4803303/smarthack-wifi.log)

kueblc, 2020-06-19:
Unfortunately this is not an ESP82xx, and thus we cannot install ESP firmware on this device

closed 2020-06-19 by kueblc

cgull, 2020-06-21:
Oh, but it is.  The FCC ID on the bulb is syw-a19rgbwagt2, and the photos at https://fccid.io/SYW-A21RGBWAGT2/Internal-Photos/Internal-Photos-4415330 show a TYWE3L, which Tuya's own website confirms is ESP8266 based.

Some further investigation shows that the bulb does a DNS lookup for a nonexistent host, and if it gets an A record, stops there.  If I configure dnsmasq to not return an A record, the bulb attempts connections to a couple of addresses owned by us-west EC2 hosts on port 20443.  I haven't tried redirecting these to tuya-convert with iptables yet, I'll try that soon.

kueblc, 2020-06-21:
This is an FCC violation for sure, they should have gotten a new FCC ID when they switched modules. But unfortunately we've been seeing this quite frequently.

OUI for `68:57:2d` is "Hangzhou Aixiangji Technology Co., Ltd"

All ESP82xx have an OUI of "Espressif Inc"

cgull, 2020-06-24:
Right you are, this is pretty clearly not an ESP8266-based bulb, and definitely not the same bulb as in the FCC registration.  I pulled the bulb apart and did try some things on the software, and here's what I know:

* I think the module is a Tuya WB2L, which is [documented](https://docs.tuya.com/en/iot/device-development/module/wifibt-dual-mode-module/wb2l-datasheet?id=K9duegc9bualu) as being based on an ARM Cortex M4.

* Otherwise, the bulb is very similar to the Feit bulb discussed [here](https://github.com/arendst/Tasmota/issues/6534#issuecomment-564765046), the boards in my bulb are just revisions of the boards in that bulb.  It's probably possible to just swap an ESP8266-based module onto it.
* Hacking on tuya-convert to listen on the 3-so-far IP addresses it tries to connect to shows that the firmware behaves very similarly to what tuya-convert already supports, but does a couple of new things:  It attempts SSL connections with a client certificate, but it offers a different client identity each time.  The first byte is set to 2, the next 16 bytes vary on each connection, and the remaining 32 bytes are the same on each connection.

I suspect it would be possible to extend tuya-convert to support these bulbs someday, but since there's no alternative firmware yet, not much point.

A couple of sample connections from smarthack-psk.log:
```text
new client on port 20443 from 10.42.42.28:49157
ID: 02a2af0fc077565d1a683cebe1ac223dcefbef146f71981deb5ec5f01fc0e9b4af56d52c8388fe2ad137353be5db10423c
Prefix: b'\xa2\xaf\x0f\xc0wV]\x1ah<\xeb\xe1\xac"=\xce'
PSK: 0a44f8f65ed6ce00561f05acc3596c313ff12b4c6aec6c77c5480ea297a2aae5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 20443 from 10.42.42.28:49158
ID: 020256e05908f20196c3338bec2dd28b6ffbef146f71981deb5ec5f01fc0e9b4af56d52c8388fe2ad137353be5db10423c
Prefix: b'\x02V\xe0Y\x08\xf2\x01\x96\xc33\x8b\xec-\xd2\x8bo'
PSK: 31676fca49e345666198b75034292c8c1f6ad14c9d5cfc7e710cf0fbfad077c5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
```

If you or anybody else is interested in this module, I could be convinced to connect a serial adapter to it and poke at it more.

ctrl50, 2020-08-25:
Hey. Here is a new Merkury White bulb that now contains the wb2l chip. Previously the merkury bulbs had the esp8266 chips. Also. the Merkury plugs now contain this unflashable chip.

Didnt think it warranted its own topic since its not flash-able and doesn't pertain to tuya-convert. So just dropping this here for future search reference.

[image]