ct-Open-Source/tuya-convert #676: Circuit Breaker MCB could not establish sslpsk socket

jrubiofrance, 2020-06-19:
Hi,
I recently bought a circuit breaker (see picture) and i tried to flash it with the latest version of tuya-convert but it fails.

[image]

I already flashed other old devices (bought in early 2019) with older versions of tuya-converter.
I do know that Tyua securized their firmware and now the fake connection is done by using TLS.

Anyway, I've tried everything based on the most recent posts I've read in github and other pages. I do nothing wrong but.... it does not work. Find here below the logs and the tcpdump recorded during my last try (renamed to .txt)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4806420/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4806421/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4806422/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4806423/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4806424/smarthack-wifi.log)
[capture.txt](https://github.com/ct-Open-Source/tuya-convert/files/4806432/capture.txt)

So, I was wondering if: 
a) this device is no supported
b) something got wrong (outdated fake certificate?) in the code

Thank you for your support

kueblc, 2020-06-19:
Unfortunately this device is not currently supported, see #483

closed 2020-06-19 by kueblc

jrubiofrance, 2020-06-20:
Hi,
Thank you for your quick reply. Is there any possibility to enlarge the list of supported devices? I mean, how can I proceed? This is not explained anywhere.

jrubiofrance, 2020-06-20:
Hello again,
Hereinafter the data gathered using tuya-cli:
[ { id: '01257235d8bfc006ccae',
    ip: '86.246.169.60',
    localKey: '65d1131ed6126701',
    name: 'HEY100' } ]

In which script should I enter the localKey to make the FOTA runs? I've reviewed each and I did not found a way to link an specific device by giving the key. am I wrong?

kueblc, 2020-06-20:
We do not know when or if these devices will ever be supported. It will require a new vulnerability to be found and exploited.

The device credentials gathered from `tuya-cli` does not change this situation.

Your only options are to flash the device by serial, wait to see if a new exploit is developed, or use the stock firmware with the vendor software or via `tuyapi`.