ct-Open-Source/tuya-convert #677: How to tell if a device is running http or https firmware

wreality, 2020-06-20:
I have found a new device that is flashable using Tuya-convert and would be happy to add it to the wiki, but I’m not sure how to make sure I add it to the correct page. 

Is there a way to tell if a device was using the http vs https firmware when it was flashed ?

kueblc, 2020-06-20:
That's great news, thanks for your contribution Brian!

You can check the logs in the `backups` folder for that device. If the device was running HTTPS firmware, you will see entries from the device in the psk log. You will also see `et=1` in the web log.

You can also tell by the user agent in the web log, and the version number in the udp log, but you should have plenty of info to work from by now.

You do make a good point that we don't have this information on the wiki. It would make a good addition, I might try adding something when I get a moment.

wreality, 2020-06-20:
Perfect.  That's exactly the hint I needed.  I compared one of the previous devices (that was on the https list) to the new one and that cleared it up.  Looks like this one is an HTTP device. (there was just a Keyboard interrupt exception in the psk log).   I'll add it to the list.  Thanks!

closed 2020-06-20 by wreality