ct-Open-Source/tuya-convert #679: OTA flashing of new Etersky WF-CS01 Curtain Switch doesn't work

dkoeux, 2020-06-21:
Error code:  **(SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC and SSL: NO_SHARED_CIPHER)**
Use new pi4 with buster lite and android SM. This works well with OUTDOOR plug: TS-EOP03-EU,
but with WF-CS01 stop pairing mode immediately when:
   _................
   SmartConfig complete.
   Resending SmartConfig Packets_

Doesnt't work with: https://github.com/ct-Open-Source/tuya-convert
and also not with: https://github.com/digiblur/tuya-convert -b development 

Firmware: ???  (can't check)
Hardware layout of switch:
      _- EU_CBL_WF_SV1
      - FR4   94VO
      - T=1.2MM
      - 20190515_

printout: **20200306**
### 
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4810334/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4810349/smarthack-wifi.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4810350/smarthack-udp.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4810351/smarthack-mqtt.log)

kueblc, 2020-06-21:
Unfortunately this is the latest patch from Tuya, see #483

closed 2020-06-21 by kueblc

mnayef95, 2020-11-10:
@dkoeux were you able to flash it using serial flashing?

If yes could you please tell me the steps?