ct-Open-Source/tuya-convert #680: Disassembly of unknown plug module ("WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware")

TRSx80, 2020-06-22:
Hai guise,

In my [earlier thread](https://github.com/ct-Open-Source/tuya-convert/issues/484) I think I had assumed at first I was dealing with RTL8710, although later in that thread myself and another guy had some indications the chip might be WinnerMicro W600...

So today I decided to explore further.  I decide to put here in separate thread until I have positively identified what chip we are working with.  But also to document disassembly, because I already made one mistake (so others won't have to).

So my plug modules are US and look like this:

[image]

You can see where I already chewed this one up a little around the ground (I think you Brits call it "earth") pin.  Which was completely unnecessary.  I think I had assumed the front cover plate was glued or microwave welded on, and that I would probably need to destroy the thing to get it open (which I was OK with) but that turned out not to be the case at all!  It simply snaps in...  I had started out doing stuff like this (DO NOT do this):

[image]

[image]

Which was completely unnecessary.  You can see I made a mess of the edge of the plug now.  I guess I will just end up using this one somewhere no one will see it...  Anyway, all you need to do is use something thin like a plastic spudger (probably preferred) or I don't have one so I used a utility knife in the thin crack between the housing and front cover plate:

[image]

The lid pops right off, and then you can see the little tabs around the edge, and no glue:

[image]

Now, I don't know what this shmoo is, it is sticky, maybe there was supposed to be glue on the lid, but they missed?  China man in factory has empathy for us hackers?  Not sure:

[image]

Anyway, now I can see what looks like some sort of module soldered to main board:

[image]

[image]

Sorry for potato quality pic, but from bottom (left) to top (right) those are:

* 3V3
* GND
* RXO
* TXO
* PB16 or maybe P816 (?)
* PB15 or P815 or P915 (?)

OK at this point the back / bottom of circuit board where the plugs come out of the shell does appear to be attached to the shell somehow, so I will keep working on that and try and post another update later.

TRSx80, 2020-06-22:
OK, this is where the destruction comes in I think.  lol

First I tried desoldering the large connections that go to the actual plug prongs.  I could not succeed that way.

[image]

Then I said "to hell with it" and just got the Dremel out.

I very carefully cut *almost* all the way through, making sure not to damage the circuit board nor module:

[image]

[image]

Then I popped the corners out with a screwdriver, revealing the module soldering:

[image]

Then I got the bright idea to try and peek behind the module.  I carefully pulled it slightly away from the case and this is what I saw:

[image]

[image]

I searched the Internet for "Tuya TW-02" which led me to [this thread](https://github.com/ct-Open-Source/tuya-convert/issues/572) right here in this very repository.  So, the saga will continue there.

So the TL;DR, if you get the error in the title, and want to find out what kind of module you have, is:

* Just use a spudger, utility knife, or etc. to pop off the cover.

* Then gently pull the module away from the housing and look closely for a part number back there.

TRSx80, 2020-06-25:
^ Making this post strictly so I can link closer to the TL;DR directly above. ^

I suppose I should close the issue, also?

Laces, 2021-01-20:
Hi, I also bought a tuya light switch but unfortunately it didn't contained an esp8226, but something else. I managed to upload MicroPython with W600 tool but I can't use the WIFI of the chip. Did you guys saw this chip before? I'm out of ideas...
[image]