ct-Open-Source/tuya-convert #682: Aoyococr X10S bricked (when trying to flash from Docker on unsupported OS)

n8henrie, 2020-06-23:
New to tuya-convert.

I just tried flashing a Aoyococr X10S using docker on a Raspberry Pi 3 running Void Linux. It didn't work. (I know that Void is not a supported platform, this was just an experiment at my own risk and if the plug is bricked because of that so be it. Just wanted to try!)

It got to the backing up firmware phase and seemed like it was working, but afterwards the switch seems almost totally non-functional. It still blinks its light -- sometimes 2 or 3 times -- when I plug it in, but that's it. It doesn't join the vtrust-flash network, and it doesn't create a vtrust-recovery network, so I don't think it got to the intermediate firmware.

Reviewing [other issues](https://github.com/ct-Open-Source/tuya-convert/issues/553) with this device it looks like I'll have to figure out how to open it up, unless there are any other ideas. If it's not joining vtrust-flash or presenting the recovery network, are there any other steps I should consider before trying to open?

Many thanks in advance for a really cool project!

n8henrie, 2020-06-23:
Of the 4-pack, the remaining 3 seem to have flashed from Raspbian. There were a few small hiccups -- none of the firmware backups worked, and one of the switches failed partway through but was able to recover on a subsequent run. So it doesn't seem to be a problem with the switches (e.g. a firmware update).

I tried re-running the broken switch under Raspbian but had no such luck (which isn't surprising, since it didn't seem to be joining the network as I described above).

kueblc, 2020-06-24:
While we haven't tested on Void Linux, it should still work, or at least not fail catastrophically.

The behavior you're describing is consistent with a device stuck on the intermediate firmware. Just to be sure, have you tried disconnecting the device from power, and only reconnecting it once the `vtrust-flash` AP is accessible?

Logs may be in the `scripts` or `backups/*` directories, although I'm not 100% sure what Docker does with these (Docker support was submitted by a contributor).

n8henrie, 2020-06-25:
Thanks for your response. Good news, I was able to follow the same process [detailed elsewhere](https://github.com/ct-Open-Source/tuya-convert/issues/553#issuecomment-624783911) to fix the plug, so I have 4/4 running Tasmota now. Very happy to see the power usage reporting works!

I suspect the issue with Void (or at least part of it) may have do to with void using `runit` instead of systemd; for example, to get the `vtrust-flash` AP to exist for more than a few seconds I had to first turn off my existing AP with `sv`; otherwise, runit seemed to restore my default AP mode.

Yes, I disconnected from power and tried numerous iterations of trying to join `vtrust-flash`, including only powering the plug after the AP was visible and phone was connected as you described, various durations of holding the button powered or unpowered, connecting the plug while holding the button, waiting 30+ minutes, (or even overnight), etc. Could never get it to connect.

Interestingly, even with the soldered connections above and directly powering from 3.3v it still did not work and still would not connect; after a number of attempts I included a bypass capacitor as recommended and it instantly worked.

So my process, start-to-finish:

1. Carefully open the backside using a box cutter (thanks to @clharting [for the tips](https://github.com/ct-Open-Source/tuya-convert/issues/553#issuecomment-648495653))
    1. Carefully wedge the blade into the middle of the flat portion of top or bottom until it starts to open a touch with light pressure
    2. Insert a small flathead screwdriver and use this to carefully spread the opening to either side where the tabs are
    3. Repeat for the other flat side
2. Solder a 30g wire to the center ("output") and one to the ground of the 3.3v regulator (thanks to @skpram219 [for the image](https://github.com/ct-Open-Source/tuya-convert/issues/553#issuecomment-624802283), very helpful)
    - Just to see, I first tried just using an alligator clip to this output, being careful there were no shorts, and another to the plug's ground, but this didn't work -- I suspect something about AC vs DC grounding?
3. Connect a 200uF capacitor (just what I had laying around) as a bypass capacitor between these two wires
4. Start tuya-convert and get the AP ready
5. Connect (center prong) "output" to 3.3v from a ft232 uart I had laying around, and ground to its ground
6. Ensure flash succeeded by looking for the `tasoma-` AP to appear on my phone
7. Reassemble the plug, ensuring the jumper lines up correctly for the button
8. Plug into AC power and make sure it still seems to work
9. Seal up with superglue

Thanks to all for responses. I'll go ahead and close the issue, for now I'd recommend others not use Void unless they are exited about the prospect of opening up the plug. Raspbian worked a treat.

closed 2020-06-25 by n8henrie

n8henrie, 2020-07-02:
Just wanted to give a little update -- I got these switches set up with HomeAssistant <- MQTT -> Tasmota and noted that the power utilization reporting does indeed work (although I had to configure the reporting interval to be much more frequent for my use case). I [found a template](https://templates.blakadder.com/aoycocr_X10S.html) that works well for me:

```json
{"NAME":"Aoycocr X10S","GPIO":[56,0,57,0,21,134,0,0,131,17,132,0,0],"FLAG":0,"BASE":45}
```

Especially given that the small form factor only takes up 1 outlet, it's rated for 15A, and they're only around $8 each, I figured it was worth loading up on another 4 before the firmware changes.

Of the second set, the first flashed fine, with the exception of the backup failing (like all the others). I decided to comment out the backup part of the script, and the 2nd and 3rd both subsequently did not complete the process on the first go. I don't know if this was due to me commenting out the backup or just random. I was able to get both of these to complete the process on a second go; it was able to locate the intermediate firmware and complete the process without a hitch. I then restored the backup part of the script (thinking that may have been the culprit -- timing issue?), but the 4th plug failed to flash and was subsequently unable to recover the process. I tried literally *hundreds* of variations of unplugging and re-plugging, with or without clicking or holding the button, before, after, and during the setup of the `vtrust-flash` AP, toggling the power a few times quickly, letting it sit powered for hours before trying, letting it sit unpowered overnight before trying... absolutely nothing.

This morning I took it apart. This time I was able to hook into the middle prong of the voltage regulator with a small probe and used my multimeter to locate a large blob of grounded solder that I grabbed with an alligator clip, so I didn't need to solder. It *still* did not work in spite of multiple tries without the capacitor. I again attached a 200uF bypass capacitor, and it *instantly* worked.

Connected back up minding the jumper, updated Tasmota and made sure everything worked, then applied a little superglue.

So after 8 switches, my experience with the Aoyococr X10S has been:

- Much of the time tuya-convert works great with minimal effort, although the firmware backup has failed every time
- About 1/3 of the time there is some kind of issue with the initial flash
  - Of these, about 1/2 of them were resolved by just re-running tuya-convert, or unplugging and replugging and then re-running tuya-convert -- usually this took no more than 1-2 minutes to make obvious progress here
  - The other 1/2 I was unable to get to work without opening the devices and powering the ESP8266 directly, which was not terribly difficult
  - IME, powering the device properly required a bypass capacitor -- I did not get it to work even once without one, and it worked instantly every time with one

Ultimately I have 8/8 switches working great, I'm very pleased. Thank you!