ct-Open-Source/tuya-convert #683: Tuya Convert not working, either with mobile 'phone or 'donor'

jimford-sudo, 2020-06-25:
Have tried many times to flash a pair of bulbs without success and have little idea of where the problem might lie. Here's the out put from smarthack-wifi.log:

Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:8b:ae:bb and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED ec:d0:9f:14:fd:46
wlan0: AP-STA-DISCONNECTED ec:d0:9f:14:fd:46
wlan0: AP-STA-CONNECTED ec:d0:9f:14:fd:46
wlan0: AP-STA-CONNECTED 2c:f4:32:3f:69:61
wlan0: AP-STA-DISCONNECTED 2c:f4:32:3f:69:61
wlan0: AP-STA-CONNECTED 2c:f4:32:3f:69:61
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 2c:f4:32:3f:69:61
wlan0: AP-STA-DISCONNECTED ec:d0:9f:14:fd:46
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

I'd be grateful for some help, please.
Jim

kueblc, 2020-06-25:
Hi Jim, we'll need more than just your wifi log to be able to help. Please share the rest of them.

jimford-sudo, 2020-06-25:
Many thanks for the reply 'kuebic'. Here you go:
## smarthack-mqtt.log ##
1593099269: mosquitto version 1.5.7 starting
1593099269: Using default config.
1593099269: Opening ipv4 listen socket on port 1883.
1593099269: Opening ipv6 listen socket on port 1883.
^C1593099367: mosquitto version 1.5.7 terminating
#
## smarthack-psk.log ##
[Last few lines, after many identical lines but with differing ports.]
new client on port 443 from 10.42.42.33:4291
ID: 0242416f68626d64366147393149465231a7cbc7912eaea2c6dfc7cd852464d1ab0e33123370c6291a671c30bc8f993a1d
PSK: e53172c96317d31d08a8cd0c7042424a7644cd2e38a2b1824552a930ab72a570
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.33:20803
ID: 0242416f68626d64366147393149465231a7cbc7912eaea2c6dfc7cd852464d1ab0e33123370c6291a671c30bc8f993a1d
PSK: e53172c96317d31d08a8cd0c7042424a7644cd2e38a2b1824552a930ab72a570
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
#
## smarthack-udp.log ##
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C
#
## smarthack-web.log ##
 Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...
#
## smarthack-wifi.log ##
System info
===========
31856ad
Linux openmediavault 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:8b:ae:bb and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 48:3f:da:53:f7:bd
wlan0: AP-STA-CONNECTED 2c:f4:32:3f:69:61
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 2c:f4:32:3f:69:61
wlan0: AP-STA-DISCONNECTED 48:3f:da:53:f7:bd
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
#
I'm running on a Raspberry Pi 4, having also tried a 3. Also tried using a Wemos D1 Mini donor, and a mobile 'phone.
Jim

kueblc, 2020-06-25:
Based on your psk log, it looks like this is issue #483, which unfortunately we don't have a resolution for currently. For now your options are to flash it by serial or use with stock firmware. We'll be sure to post in #483 if anything changes.

closed 2020-06-25 by kueblc

jimford-sudo, 2020-06-25:
Many thanks 'kueblc' for casting an expert eye over my logs. I don't know what flashing by serial is - I'll do some research. I'll also keep an eye on #408  Best wishes, Jim