ct-Open-Source/tuya-convert #684: could not establish ssl psk socket

rhamblen, 2020-06-27:
OK so I just got some eLinkSmart Wifi socket plugs (UK) from Amazon.  They look like a an OEM version of the Teckin SP27.  They are 16A and they have a mac address corrisponding to Espressif and are identified as an ESP_xxxxxxx host name when I connect to it.   
So I went to use Tuya Convert and the wifi log sees me connect to the device.
Nothing is happening on the  MQTT log - just opens listening sockets
The udp log sees my Teckin SP23 (new type so cant convert) but doesnt see the new sockets.
Web log is just listening.

I can see the phone in the psk log
I can see the wifi log connect to  the new device
the udp log then sees the new device but fails to setablist sslpsk socket:[SSK: DECRYTION_FAILED_OR_BAD_RECORD_MAC] (_ssl.c:1056)

I can send the logs if required, but I cant see anything obvious wrong here.  The device should be able to be flashed as it is an ESP devicie.

Also, one device was connected to the tuya cloud so I could double check mac addresses and the other wasnt.  They both have same problem.

mdebus, 2020-06-28:
Same problem here. I got a Teckin SS42 outdoor socket.

rhamblen, 2020-06-28:
its the psk file which is interesting!

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4842350/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4842351/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4842352/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4842353/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4842354/smarthack-psk.log)

kueblc, 2020-06-28:
Duplicate of #483

closed 2020-06-28 by kueblc