ct-Open-Source/tuya-convert #685: Zemismart YH-002 Blinds Controller

richard-parker, 2020-06-30:
Hey,

Firstly, flashed a ton of devices very successfully with Tuya-Convert, so thanks for the great work there on this project.

I'm not having much luck getting my Zemismart YH-002 blinds controller to work though. To my understanding this is a Tuya device. But, it seems to only have the kind of pairing mode where the device presents its *own* AP, eg. "SmartLife-XXXX". I'm not sure if Tuya-Convert provides a mechanism to flash devices in that particular pairing mode (I don't think so?), and I'm kinda at a loss for how to flash this particular device.

I have seen over on the Tasmota forums that some folks have clearly used Tuya-Convert to successfully flash theirs, but the process they went through isn't described. I wondered if anyone here could provide any hints?

Tried a bunch of key combinations, I get the "fast-blink" LED flash, but still no joy.

Thanks

kueblc, 2020-06-30:
tuya-convert only supports the "fast-blink" pairing mechanism, not AP mode.

Check your logs for more information. Most common issue these days is #483, so check your psk log for a PSK ID starting with `02`. If so, unfortunately tuya-convert is not possible on this firmware. Otherwise post your logs and I'll take a look.

richard-parker, 2020-06-30:
> tuya-convert only supports the "fast-blink" pairing mechanism, not AP mode.
>
> Check your logs for more information. Most common issue these days is #483, so check your psk log for a PSK ID starting [...]

Thank you: I will pull the logs off soon and hunt for the PSK ID starting with 02, otherwise I will post the logs. I’ll close the issue if it turns out it is related to #483. Thank you.

richard-parker, 2020-06-30:
@kueblc you were correct, **smarthack-psk.log** displays:

```
new client on port 443 from 10.42.42.30:9775
ID: 0242416f68626d64366147393149465231ad98b470a18e1067347453ef9b07fc216f7ee4353$
PSK: fb5a4ba4e5975e2216b86e013676771d0c0f8248e33af96df46d5d609348d7c5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] d$
```

... which I guess means it's the newer firmware and thus not compatible with TuyaConvert.

Thanks for the super-fast response. I'll close the issue here but preserve the full logs; if any of these would be useful to the community let me know as I'm happy to share those. Though I see logs do not seem like they're in short supply based on the comments in #483 !

closed 2020-06-30 by richard-parker