ct-Open-Source/tuya-convert #686: Can't select custom .bin file

Draxter, 2020-07-01:
Hi, I'm trying to flash the BHT-002-GCLW Thermostat with this custom binary: https://github.com/klausahrenberg/WThermostatBeca/blob/master/WThermostat_1.11.bin

I copied the binary into /tuya-convert/files/WThermostat_1.11.bin as explained on the docs page here: https://github.com/ct-Open-Source/tuya-convert#flash-third-party-firmware

But it never appeared on the list of choices to flash:

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing

So I chose 0. Any idea what am I doing wrong?

Logs:
[smarthack-mqtt](https://pastebin.com/0GVc1xjr)
[smarthack-psk](https://pastebin.com/5er0ntwL)
[smarthack-web](https://pastebin.com/exHNiBXR)
[smarthack-wifi](https://pastebin.com/cQ3kVQk8)
Smarthack-udp:
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
192.168.1.9 {"ip":"192.168.1.9","gwId":"47163522cc50e3e660c8","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"IAYz2WK1th0cMLmL","version":"3.3"}
10.42.42.40 {"ip":"10.42.42.40","gwId":"3157168698f4abc2a2ec","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"IAYz2WK1th0cMLmL","version":"3.3"}

kueblc, 2020-07-01:
The firmware must meet certain conditions outlined in the README. It must be smaller than 512kb and it must include the bootloader. If it does not meet these requirements, it will not be displayed as an option as a basic safety mechanism.

Draxter, 2020-07-01:
Thanks for a quick reply. It's way below 512kB, is a bootloader a separate file? Thsi is the repo in question btw:
https://github.com/klausahrenberg/WThermostatBeca

kueblc, 2020-07-01:
The bootloader is the first sector of the binary, beginning with byte `E9`. The binary in the repository you linked looks fine to me, and indeed it shows up on my firmware selection menu.

You placed the `.bin` inside the `files` directory, along side the existing `tasmota` and `espurna` binaries, correct?

Draxter, 2020-07-01:
Yes, that’s right. Very strange.  It also had the same permissions and ownership as other binaries

Draxter, 2020-07-01:
My bad, I ran 'wget https://github.com/klausahrenberg/WThermostatBeca/blob/master/WThermostat_1.11.bin' which ended up downloading some gitlab's HTML file and named it WThermostat_1.11.bin. 😅

closed 2020-07-01 by Draxter