ct-Open-Source/tuya-convert #687: Problem flashing KMC 30137

ggodkin, 2020-07-02:
Error log suggests that device is not ESP82xx,  however FCC ID 2AFNL-TYWE3S indicates that WiFi device is 8266. It seems like Tuya is fighting back by exploiting following code:
			if "ablilty" in data:
				print("WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware")
		except:
			pass
I have few devices from the same pack flashed successfully and few failed all based on MAC address. Successful had MAC 24:62:AB:xxxxxx and failed 68:57:2d:xxxxx
The log shows:
wlx00a1b0237a1c: AP-STA-CONNECTED 68:57:2d:9f:9c:d2^M
10.42.42.30 {"ip":"10.42.42.30","gwId":"b98a820b62b53448","active":0,"ablilty":0,"encrypt":true,"productKey":"keynwa8qwvn4uwkf","version":"3.3"}^M
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware^M
^C1593657688: mosquitto version 1.4.8 terminating^M
^CReceived SIGINT, exiting...^M
^Cwlx00a1b0237a1c: interface state ENABLED->DISABLED^M
wlx00a1b0237a1c: AP-STA-DISCONNECTED 68:57:2d:9f:9c:d2

kueblc, 2020-07-02:
FCC ID is not a reliable method for determining chipset. Manufacturers have been willfully violating FCC by changing chips without changing the ID. Unfortunately this is another example. The `68:57:2d` OUI is not an ESP.

kueblc, 2020-07-02:
Regarding exploiting our detection mechanism, it would make no difference. Our detection mechanisms only exist to help debug, and have so far proved reliable. They do not alter the tuya-convert process in any way. You'll see that the code branch you quoted does nothing but print a message.

closed 2020-07-02 by kueblc

divadiow, 2026-02-02:
sure. join Elektroda and make a post? 

https://github.com/divadiow/BK7231GUIFlashTool also supports RDA5981 read/write