ct-Open-Source/tuya-convert #688: Aoycocr X10S Bricked?

cblt2l, 2020-07-02:
I have 4 smart plugs brand Aoycocr X10S

I was able to successfully flash and run tasmota on 2 of them, however I believe the other 2 may be bricked.

During the flashing process for the bricked ones the backup failed so I canceled & restarted the process. After this when I plug them in, the light slowly blinks (about 1 blink per second) and pressing the button does nothing. Holding the button also does nothing. The light will continue to flash for a few minutes and then stop. There is no wifi network created by the plug that I can see. From the logs it appears they are not connecting to the vtrust-flash network. I can not tell if the intermediate firmware is installed or not.

On the ones that flashed without issue, once the backup failed I just continued on and everything else went smooth. These plugs run tasmota with no issues.

I'm running everything in a docker container on my Ubuntu 20.04 machine with a Intel 9260 wifi card.

Any ideas?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4866224/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4866225/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4866226/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4866227/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4866228/smarthack-wifi.log)

kueblc, 2020-07-02:
See #682

cblt2l, 2020-07-02:
Thanks, did not see that one. I will start there!

cblt2l, 2020-07-19:
Just to follow up,

I followed the instructions in [#682 ](https://github.com/ct-Open-Source/tuya-convert/issues/682) using a 100 uF capacitor and an arduino uno for 3.3V and was able to flash both of the non-responsive plugs without issue. 

Thanks everyone!

closed 2020-07-19 by cblt2l