ct-Open-Source/tuya-convert #689: Unable to flash VBS LED bulb

taha-yassine, 2020-07-02:
Hi everyone,

I'm new to tuya-convert and I bought [this](https://fr.aliexpress.com/item/4000290763952.html?spm=a2g0o.productlist.0.0.58d340465XsOGl&algo_pvid=a90e398c-58bb-4649-a419-6ff4b1d3fae3&algo_expid=a90e398c-58bb-4649-a419-6ff4b1d3fae3-56&btsid=0b0a0ad815937294765332054e0b05&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) VBS (??) LED light bulb from aliexpress. I tried to flash it with no success. I don't know if I'm doing something wrong and I don't know if this is related to the new PSK format.

Here are the logs:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4866928/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4866930/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4866931/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4866932/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4866933/smarthack-wifi.log)

Thanks.

kueblc, 2020-07-03:
I see only one device connected in your wifi log. Is that your phone?

Did you put the light into fast blink mode before pressing enter?

taha-yassine, 2020-07-04:
Okay so I was doing something wrong. I was putting the light into slow blink mode. So I managed to put it in fast blink mode but I still have the same problem. Only my phone shows up as a connected device (what appears in the logs). Seems like for whatever reason the light goes undetected.

taha-yassine, 2020-07-04:
I think I found the issue. The light bulb is simply not using an ESP8266 but rather what seems like an RDA5981 as explained in [step 2 of this article](https://www.instructables.com/id/Hardware-and-Software-Hack-Smart-Devices-Tuya-and-/). I guess I got unlucky.

P.S. If anyone knows of a way to flash this module with a custom firmware, I would be very grateful.