ct-Open-Source/tuya-convert #691: Docker Container, can't connect to vtrust-flash

kkniffin, 2020-07-06:
I followed the instructions in the guide to run the docker container. however, I can't seem to get the vtrust-flash ssid to lease out an IP address. When I connect my phone/laptop it just goes from connecting to no longer connected. Am i missing something obvious?
 ```
System info
===========
94acc3b
Linux crypto-VirtualBox 5.3.0-62-generic #56-Ubuntu SMP Tue Jun 23 11:20:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
OpenSSL 1.1.1c  28 May 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.9
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.5
===========
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlx9cefd5fcea80...
Configuration file: /dev/stdin
Using interface wlx9cefd5fcea80 with hwaddr 9c:ef:d5:fc:ea:80 and ssid "vtrust-flash"
wlx9cefd5fcea80: interface state UNINITIALIZED->ENABLED
wlx9cefd5fcea80: AP-ENABLED 

==> /opt/tuya-convert/scripts/smarthack-web.log <==
Listening on 10.42.42.1:80

==> /opt/tuya-convert/scripts/smarthack-mqtt.log <==
1594066225: mosquitto version 1.6.6 starting
1594066225: Using default config.
1594066225: Opening ipv4 listen socket on port 1883.
1594066225: Opening ipv6 listen socket on port 1883.

==> /opt/tuya-convert/scripts/smarthack-udp.log <==
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
```

dgrostoto, 2020-07-12:
Same problem for me using docker.
Tried various flavor of linux (mint 18 19, with AP capable wifi interface) and raspberri pi 3B  raspberry OS buster.
Same problem each time : "RTNETLINK answers: File exists" as describer above.

kueblc, 2020-07-14:
Don't worry about

> RTNETLINK answers: File exists

It is not a problem.

You mention it won't lease an IP address, can you confirm that `dnsmasq` is running and the DHCP port is open?

kkniffin, 2020-07-14:
Seems to be:

nobody     263  0.0  0.0  53168   380 ?        S    18:10   0:00 dnsmasq --no-resolv --interface=wlx9cefd5fcea80 --bind-interfaces --listen-address=10
.42.42.1 --dhcp-range=10.42.42.10,10.42.42.40,12h --address=/#/10.42.42.1

erdoking, 2020-08-09:
Stop the NetworkManager ... He seems to reactivate the "normal" wifi configuration during flash

`sudo systemctl stop NetworkManager`

dgrostoto, 2020-08-16:
Raspberry OS does not embed a NetworkManager. This is not the source of the problem.

tablatronix, 2020-11-24:
Similar issue on raspi 4, docker compose, I am not seeing any logs though, is there a special branch or image I should be using or flag, if not my volumes must be borked

tablatronix, 2020-11-26:
[image]

commit 71aaa1fb (2020-12-02): Merge pull request #821 from lukavia/master

Minor fixes for docker execution. Fixes #691
closed 2021-01-26 by kueblc