ct-Open-Source/tuya-convert #692: Merkury MI-BW942-999w bulb stuck in smartconfig

Didgeridrew, 2020-07-08:
This is my first time using tuya-convert and I've seen a few websites where other people say they have been able to get these bulbs to flash, but I've tried multiple bulbs and they all get stuck.

I'm using a RPi Zero W set up as a serial device as described in the wiki.
I have tried a phone, a tablet, and an ESP32 as the donor device to connect to vtrust-flash.  All of them seem to connect without any issues.

Terminal 

``` 
...
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
            

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
.Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]
```

smarthack-wifi.log

``` 
System info

===========

94acc3b

Linux raspberrypi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux

OpenSSL 1.1.1d  10 Sep 2019

Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley

Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP con$

This software comes with ABSOLUTELY NO WARRANTY.

Dnsmasq is free software, and you are welcome to redistribute it

under the terms of the GNU General Public License, version 2 or 3.

hostapd v2.8-devel

User space daemon for IEEE 802.11 AP management,

IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator

Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors

Python 3.7.3

===========

Configuring AP interface…

RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:13:06:2e and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 3c:71:bf:fe:50:2c
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

smarthack-web.log

``` 
Listening on 10.42.42.1:80

^CReceived SIGINT, exiting... ```

smarthack-psk.log

``` ^CTraceback (most recent call last):

  File "./psk-frontend.py", line 113, in <module>

    main()

  File "./psk-frontend.py", line 106, in main

    r,_,_ =  select.select(readables, [], [])

KeyboardInterrupt
```

smarthack-mqtt.log

```
1594222799: mosquitto version 1.5.7 starting
1594222799: Using default config.
1594222799: Opening ipv4 listen socket on port 1883.
1594222799: Opening ipv6 listen socket on port 1883.
^C1594222991: mosquitto version 1.5.7 terminating
```

smarthack-udp.log

```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667

^C
```

kueblc, 2020-07-09:
I see an ESP device in your wifi log

> wlan0: AP-STA-CONNECTED 3c:71:bf:fe:50:2c

Is this your smart device or the donor?

Didgeridrew, 2020-07-09:
Yes

justinmartin, 2020-07-11:
I'm having the same problem. I assume its related to #483

jonaseymour, 2020-07-12:
Yeah, I think I'm having a similar issue. My RP 3B is creating the network `vtrust-flash` but with no password required.
wifi.log
```
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:ad:ab:62 and ssid "vtrust-flash"`

```

mati22124, 2020-07-13:
Hi!

 Bought this same bulb but did. Anyone check if this has an ESP based chip. I know the one with the model number MI-BW902-999W has an esp based chip. Is this one good. I dont know how to open my bulb (nor do I want to) but could somebody verify this?

Didgeridrew, 2020-07-13:
Decided to take the plunge and sacrifice one of the bulbs this morning, to see if maybe they changed the mc.  The chip on the wifi module is marked "BEKEN BK7231 10N32"

mati22124, 2020-07-13:
If it had a different chip, wouldn't it say something along the lines of This bulb dosen't have a ESP82xx chip therefore its not flashable?

mati22124, 2020-07-13:
[image]

Does the WiFi module look like this? Module high listed in red.

mati22124, 2020-07-14:
@justinmartin I basically Have the same psk logs as you. I wonder why. They aren't the same logs as issue #483.

Didgeridrew, 2020-07-14:
My pcb is a totally different layout... and the mc had a shield over it that I had to pry off to get the info.
[image] (Sorry about the quality, the only way I could get a pic quickly was to set it on the scanner)

mati22124, 2020-07-14:
Hi! @Didgeridrew. The esp is the white chip in the upper right corner. That is what will get re flashed. At least we can confirm that this is the esp chip. If you look closely there is a small antenna at the top. What you read on the board might be a processer of something like that. In my case, I could the Mac address of the bulb by connecting it to the app. It came with updated firmware :-( v1.03. But in my case the bulb connects go the crust-flash network. I wonder whats happening?

kueblc, 2020-07-14:
> Decided to take the plunge and sacrifice one of the bulbs this morning, to see if maybe they changed the mc. The chip on the wifi module is marked "BEKEN BK7231 10N32", so (as I understand it) that's the issue.

You are correct, this is unfortunately not an ESP, this is [Alibaba's IoT SoC](https://github.com/bekencorp/bk7231_alios_sdk).

@justinmartin that does not look like #483, you would see `PSK ID 02....`.

@jonaseymour the `vtrust-flash` AP should not have a password, this is intentional. However the following line is strange, considering you said the AP is visible:

> Failed to create interface mon.wlan0: -95 (Operation not supported)

> If it had a different chip, wouldn't it say something along the lines of This bulb dosen't have a ESP82xx chip therefore its not flashable? Over here it says Device did not appear with the intermediate firmware. Could you send pictures of the inside?

@mati22124 not necessarily, we try to detect in a few situations where a device connects but is not running an ESP, but if the device does not connect there's no way we can tell.

Closing as the original issue has been found (@Didgeridrew's bulb is not using an ESP). Feel free to continue discussion however.

closed 2020-07-14 by kueblc

mati22124, 2020-07-17:
Is this an esp chip? If I opened this up to flash it manually will I be able to get tasmota on it?

kueblc, 2020-07-17:
@mati22124 the picture you posted looks like a TYLC2, which is an ESP based Tuya module, so in your case you should be able to flash by serial and use Tasmota.

mati22124, 2020-07-17:
@kueblc Sadly that was a picture I pulled of the internet

jonaseymour, 2020-07-20:
> You are correct, this is unfortunately not an ESP, this is [Alibaba's IoT SoC](https://github.com/bekencorp/bk7231_alios_sdk).
>
> @justinmartin that does not look like #483, you would see `PSK ID [...]

Apologies, this was actually an issue with a [different bulb](https://github.com/ct-Open-Source/tuya-convert/issues/708). I was having what seemed like the same issue, so posted here without thinking...