ct-Open-Source/tuya-convert #694: Ajax Online Vintage Smart Bulb Won't Flash

ijwitw, 2020-07-09:
Hi,

I tried flashing this from a Kali Linux Live CD, connecting my Pixel 2XL to the AP first (also tried another laptop, although I don't understand this part of the process anyway).

The bulb stops flashing and lights up solidly as if it has connected, but tuya-convert repeatedly reports "resending SmartConfig packets" and eventually fails.

Log files attached, any help appreciated. Thanks

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4896219/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4896220/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4896221/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4896222/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4896223/smarthack-wifi.log)

kueblc, 2020-07-09:
Duplicate of #483

closed 2020-07-09 by kueblc

timecop97, 2020-07-10:
@kueblc I had the same issue with my Ajax vintage bulb, I reached the same conclusion that the psk logs indicated it was running the new authentication, glad to have it confirmed. How would i best keep up to date on the progress regarding the authentication? Can I help out in any way?

kueblc, 2020-07-10:
You can subscribe to issue #483, any progress or ways to help will be discussed there.