ct-Open-Source/tuya-convert #696: Recover to stock firmware

jdedfamily, 2020-07-12:
Hi ,

I have successfully flashed a Arlec Power board , turns out one of the power ports is faulty . To return it under warranty I will need to recover to the stock firmware

 I have the stock firmware in my backup folder but cannot seem to find the documentation on how to do this using Tuya convert. 

The power board is currently running the latest tasmoto firmware.  8.3.1(tasmota)

kueblc, 2020-07-14:
Unfortunately you cannot restore the original firmware OTA, you would need to open the device to flash the backup over serial.

nospam2k, 2020-12-17:
Appologies if I missed this, but shouldn't this be documented that you cannot go back to stock without dissassembly? I wish I would have know this before I flashed them. In the mean time, does anyone have a procedure to remove the LED board without damaging the housing?

CarlosGS, 2020-12-17:
Ouch, I agree, this should be clarified in the instructions. Here are a couple pictures of the Tuya Bulb teardown:

The outer part is glued. Hold firmly with rubber gloves and open it with a "bending" action:
[image]
(Later it should snap back in place without leaving any visible damage, you could even add some silicone for full "original" feel. **However an expert could notice if they looked inside, keep reading**)

The internal dome is also glued, though it's trickier to remove. Gently with a flat screwdriver & be careful not to damage the aluminium PCB (specially near the two resistors). Then remove three screws, gently pull out the board and you're in:
[image]
(It has some thermal compound, handle carefully to keep it in place. **For reassembly the internal dome needs to be glued back in place with silicone or epoxy, so I doubt it can ever look original again**)

At this point there is precarious access to the programming pads, you can see them on the white PCB inside. It is quite tricky but they can be soldered with wires, flashed with any serial programmer, and desoldered back. (including IO0 which should be tied to GND to set the ESP8266 into programming mode)
[image]
I wasn't able to tear them down any further.

*Personal note: Apart from this, I'm very happy with these bulbs, they are pretty neat. I had to open them because they were running the latest Tuya firmware, which is ciphered and blocks "tuya convert". I'm running them with ESPHome and setting a maximum PWM of 40% -that's >80% perceived brightness-, so they barely get warm and should last a lifetime. Let me know if you plan on using them with Home Assistant to send you the ESPHome template.*

In any case I'm sorry this info came late to you as well. If only these were all of our problems! :)