ct-Open-Source/tuya-convert #697: Unable to flash BPA800 Feit Bulb

rezadast2, 2020-07-12:
**Are you reporting a successful flash?**
No

**Do you have a question about configuring your newly flashed device?**
No

**Did you have issues getting a device to flash?**
 BPA800/RGBW/AG/2(P) BPA800/RGBW/AG/2(P)

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/4907944/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/4907945/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/4907946/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/4907947/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/4907948/smarthack-wifi.log)

Got the bulb in fast blinking mode. Hit enter on tuya-conver start script with my phone connected to the vtrust ssid.
Blub went solid and got this error.
```
SmartConfig complete.
Auto retry in 2s. ................................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
```

mati22124, 2020-07-13:
I’m having the same problem except on the merkury bulb. I will tell you if I find a solution

rezadast2, 2020-07-13:
So i just tried it again for giggles. This time instead of the light being 20 feet away, I placed it right next to the raspberry pi and it worked.

closed 2020-07-13 by rezadast2