ct-Open-Source/tuya-convert #698: Koogeek Smart Plug #KLUP1  {Not Flashing}

Gerry4321, 2020-07-14:
Hello,
Device - "https://www.amazon.co.uk/gp/product/B07XGRG9SB/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1"
Probem - Device failed to flash using Raspberry Pi 3 - Model B (Linux raspberrypi 4.19.66-v7+)
Smart Plug in pairing mode with rapid flashing LED.

Method 1, with Wemos d1 as AP (device LED flashing, then solid as flashing commences)
Log:
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: Network is down
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:79:30:e5 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED dc:4f:22:17:f3:6c
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED dc:4f:22:17:f3:6c
wlan0: AP-DISABLED
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

Method 2, with android phone as AP (phone gets ip from pi)
Log:
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:79:30:e5 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 2c:fd:ab:ce:ce:3d
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 2c:fd:ab:ce:ce:3d
wlan0: AP-DISABLED
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

Please help. Thank You.

kueblc, 2020-07-14:
Please post the rest of your logs, there's not much to go on here

Gerry4321, 2020-07-15:
{log purged, restart, using wemos d1 (solid LED)}

System info
===========
94acc3b
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
OpenSSL 1.1.0l  10 Sep 2019
Dnsmasq version 2.76  Copyright (c) 2000-2016 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-$

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.6
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2016, Jouni Malinen <j@w1.fi> and contributors
Python 3.5.3
===========
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: Network is down
RTNETLINK answers: Network is down
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:79:30:e5 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED dc:4f:22:17:f3:6c
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED dc:4f:22:17:f3:6c
wlan0: AP-DISABLED
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

Gerry4321, 2020-07-15:
sudo nano smarthack-psk.log

^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

sudo nano smarthack-udp.log

Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C

sudo nano smarthack-web.log

Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...

Gerry4321, 2020-07-20:
Any advice ?

kueblc, 2020-07-21:
I don't see your smart plug appear in the logs. It may not be an ESP8266 based Tuya module.

Are you able to retrieve the MAC address of the unit? You can then check to see if the OUI is Espressif Inc.

Gerry4321, 2020-07-24:
Yes its Espressif Inc...

kueblc, 2020-07-25:
Does the device stop blinking when you are running `start_flash`?

nogueirasa, 2020-07-27:
Hello @kueblc , I have the same problem and the device was blinking fast (EZ Mode) and after start_flash the device stop blink.