ct-Open-Source/tuya-convert #700: Issue Flashing new Version of Novostella LED Flood Light

Schlabberanesi, 2020-07-15:
I succesfully flashed such lights 2 weeks ago. This time, it doesnt work. The light stops blinking, but nothing changes. It might be the same issue as in https://github.com/ct-Open-Source/tuya-convert/issues/483

PSK shows the following:
ID: 0242416f68626d6436614739314946523173106c1b5b6c27c3886116567b35e847ffa4b2ab4ab64e1704d1f260d7b85a31
PSK: 165bbd09dc3281cb53b181846b026bc958b4283dee0dbdb43341b1fc155d34b5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad$

I added all logs bellow. I dont know what to do next - help would be great!

pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This i            s usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
[... 118 lines cut ...]
new client on port 443 from 10.42.42.25:57213
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.25:57214
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.25:57218
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.25:57219
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.25:57220
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.36:32635
ID: 0242416f68626d6436614739314946523173106c1b5b6c27c3886116567b35e847ffa4b2ab4ab64e1704d1f260d7b85a31
PSK: 165bbd09dc3281cb53b181846b026bc958b4283dee0dbdb43341b1fc155d34b5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad$
new client on port 443 from 10.42.42.36:30070
ID: 0242416f68626d6436614739314946523173106c1b5b6c27c3886116567b35e847ffa4b2ab4ab64e1704d1f260d7b85a31
PSK: 165bbd09dc3281cb53b181846b026bc958b4283dee0dbdb43341b1fc155d34b5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad$
new client on port 443 from 10.42.42.36:2563
ID: 0242416f68626d6436614739314946523173106c1b5b6c27c3886116567b35e847ffa4b2ab4ab64e1704d1f260d7b85a31

Schlabberanesi, 2020-07-15:
IP 10.42.42.25 is my phone, 10.42.42.36 is the light in question

kueblc, 2020-07-15:
Currently not possible to flash OTA. See #483

closed 2020-07-15 by kueblc

ethank, 2020-07-16:
I love these lights. I bought 6 and flashed them successfully. Where'd you buy them from where the firmware was changed to not support OTA? I might buy some to just convert them for the future.

Schlabberanesi, 2020-07-23:
From Amazon DE

tt-appdev, 2020-07-24:
@kueblc Having the same issue. I've seen you suggest using Wireshark while configuring with API, do you have general instruction for this or any other advice for capturing the necessary data? Im an application developer by day and its been years since I've had to use Wireshark but happy to help in any way possible

TATERxxWARDY, 2020-07-24:
@kueblc Having the same issue. I've seen you suggest using Wireshark while configuring with API, do you have general instruction for this or any other advice for capturing the necessary data? Im an application developer by day and its been years since I've had to use Wireshark but happy to help in any way possible

kueblc, 2020-07-25:
Hi @TATERxxWARDY, I appreciate your willingness to help. We currently have more than enough data to look at right now, unless you happen to have a device of the same model with older firmware.

TATERxxWARDY, 2020-07-26:
@kueblc I do not, sorry