ct-Open-Source/tuya-convert #703: Ce Smart Home outlet - Can't flash

lemaygabriel, 2020-07-17:
I have two Ce Smart Outlet that I can't flash. 
[image]
I always get the message after many retries that that it can't find the device. I have a D1 mini as a donor device with the solid LED. Tries many time with different sequences of ''config mode'' on the device. The firmware version of the device is 1.1.5. 

I do not know how to get the logs

Thanks ! :)

kueblc, 2020-07-17:
The logs are stored under the `scripts` directory

lemaygabriel, 2020-07-17:
What is the command line to enter to get them ?

Thanks ! :)

kueblc, 2020-07-17:
Try `cat scripts/*.log`

Tipalo, 2020-08-01:
Any updates?
These are on the supported device list. I have the newer power monitoring version of this and was able to flash them.  They are now running 8.4.0

dlabrie, 2020-11-22:
1606006285: mosquitto version 1.5.7 starting
1606006285: Using default config.
1606006285: Opening ipv4 listen socket on port 1883.
1606006285: Opening ipv6 listen socket on port 1883.
^C1606006408: mosquitto version 1.5.7 terminating
new client on port 443 from 10.42.42.15:20278
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:48771
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:12743
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:3839
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:14954
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:16467
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:23538
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:3181
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:33508
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:20442
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:274
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:5075
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:53262
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:17056
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:9908
ID: 0242416f68626d64366147393149465231af9dad0cf99690d856f4eeb9782ec074565094db8bfe188fd0352607b606704d
PSK: faf44d30725a169b4834686a42ae16aa36d51808097601849f099e7be0562750
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.15:65160
[... cut ...]
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 68:57:2d:c8:94:3d
wlan0: AP-STA-CONNECTED c4:4f:33:af:30:25
wlan0: AP-STA-CONNECTED f8:10:93:eb:30:bd
wlan0: AP-STA-DISCONNECTED f8:10:93:eb:30:bd
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED c4:4f:33:af:30:25
wlan0: AP-STA-DISCONNECTED 68:57:2d:c8:94:3d
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

Looks like this is due to an updated firmware.