ct-Open-Source/tuya-convert #705: Neo Coolcam NAS-WR01W-16A stuck on intermediate firmware

lerra, 2020-07-17:
Hi,

I while ago I used tuya-convert successfully on 2/3 neo coolcam's, the one left out I thought got bricked. Now six months later when I looked more into the problem, I see now it is stuck on the intermediate firmware.

I have been trying to check the wiki and earlier issues without understanding what my next step is. I tried the npm package (from https://github.com/ct-Open-Source/tuya-convert/issues/472) but got the following error:
{ Error: connect ECONNREFUSED 192.168.4.1:6668
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1113:14)
  errno: 'ECONNREFUSED',
  code: 'ECONNREFUSED',
  syscall: 'connect',
[logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/4937242/logs.zip)

  address: '192.168.4.1',
  port: 6668 }

Logs attached. any help?

kueblc, 2020-07-17:
From your web log, it looks like the device did grab Tasmota. Are you sure it is not running Tasmota now?

You can try to continue the process by resuming `start_flash`, but you do not need to connect another phone if the device is already running the intermediate firmware. It will also not respond to `tuyapi` with the intermediate firmware, as `tuyapi` only works with Tuya firmware.

lerra, 2020-07-30:
Sorry for slow response,

No success :( 

I have tried with having the plug running while I do the start_flash.sh (with no mobile) and pressing enter and the same process by restarting the plug.

I also tried now with the latest checkout from git

attaching logs and output, [logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/5001041/logs.zip)

kueblc, 2020-07-30:
Nothing connected in these most recent logs, so I don't have any advice for you at the moment

lerra, 2020-07-31:
To bad,

I noticed one thing, it does announce a wifi network named ESP-XXX,

Anything that could help?

lerra, 2020-08-24:
Sorry for slow response, there is zero services/ports open on it so its not possible

dgrostoto, 2020-09-05:
Hello, I got the exactly same problem on the same plug. The only difference is that the wifi AP ESP-xxxx does not show up each time i power the plug. And when it shows up it disappear most of the time few seconds later. I suspect there is an hardware problem with the plug. For precision i bought 6 same plugs, and only one got this problem as far as now.

pizzabakker, 2020-10-06:
Hello, I got the exactly same problem with neo Coolcam Plugs. 
Last 4 i ordered all had new firmware blocking Tuya convert. :-(
Do not want my devices to be in the Tuya 'prison'.

Gunner251192, 2020-11-20:
Hi, i got the same issue with my 10!  NAS-WR01W 16A plugs. Would be really great to get this fixed.
To help i uploaded my logs when running Tuya Convert, if there is someone who can point me in the right direction it would be really great. Thanks in advance.
[NAS-WR01W.zip](https://github.com/ct-Open-Source/tuya-convert/files/5575765/NAS-WR01W.zip)

kueblc, 2020-11-20:
Hey @Gunner251192, your log file upload is password protected. I'll take a look for you when you reupload the logs. In the future, please refrain from cross posting your issue, I'll get notified when you post on any issue so once is sufficient.

Gunner251192, 2020-11-20:
> Hey @Gunner251192, your log file upload is password protected. I'll take a look for you when you reupload the logs. In the future, please refrain from cross posting your issue, I'll get notified when you post on any issue so once is sufficient.

Hi @kueblc, Wow that's a fast response :)
[NAS-WR01W.zip](https://github.com/ct-Open-Source/tuya-convert/files/5576089/NAS-WR01W.zip)
I'm quite curious

Have a nice weekend.

kueblc, 2020-11-20:
The bad news is this device is running the new patched firmware from Tuya, incompatible with tuya-convert. See #483

The good news is this device is still using an ESP82xx chip compatible with alternative firmware such as Tasmota. You can flash these "the old fashion way" by connecting to the serial port inside the device.

Gunner251192, 2020-11-21:
Yeah i know this is al solution don this in the past. But it is always a hassle and easy to break stuff. I see this is a problem with more and more devices, are there any expectations to get this working again for new devices any time soon or is that dificult to say ?

Then i can start thinking to do this manually over the serial interface or wait for the (hopefully) new Tuya Convert.