ct-Open-Source/tuya-convert #707: Feature request/Sugestion

tonyfitzs, 2020-07-19:
First let me say, I really love what you have done here fantastic project great work. 
But it’s worth noting the hardware is built to a budget and can sometimes have unknown faults. Because the process recommends NOT initially connecting the unit to the Tuya app as this may make it impossible to update, you don’t discover any issues until it’s too late and we installed new firmware.

The problem then is you can not return the device under warrantee because it no longer has the original firmware. So you need to roll the firmware back first. The only method I have seen to do this is not helpful in the use case I am describing because it relies on the user rolling back the original dumped firmware via a serial connection.

The challenge here is that not all the Tuya devices allow you to do this easily, so you may need to modify the internal circuit board (at least solder some connections) to achieve this. The problem here is that just as upgrading the firmware makes it challenging to return, modifying the hardware is even worse. 

So this got me thinking. It would be great if there were a way to install OTA a firmware stub that doesn’t do anything other than allow tuya-convert to connect to the device.

So what I mean in more detail. After you have updated the device to Tasmota or ESPurna, you use there OTA support to load a firmware shell onto the device that just turns on the WIFI pair mode the same as if it was getting ready to receive the firmware from tuya-convert. 

Then in tuya-convert, you connect to the device but rather than deploy Tasmota or ESPurna you use the original dump file back to the device. This would effectively enable the capability to re-flash the OG firmware OTA thereby solve the challenge of what to do when you need to roll back the firmware for any reason. 

I realize you don’t own the firmware involved here, so this would need to be a collaboration of some sort. But I’m sure as there is so much support for the process you have created, that this could be achieved. I will send a similar feature request to Tasmota and ESPurna.

I would love to help out in any way I can, but I would not have a clue where to start. Thanks for your indulgence.

kueblc, 2020-07-20:
Unfortunately loading the original firmware backup OTA is not possible due to memory constraints.

ESP flash memory is divided into two user partitions (and several system ones), only one of which is executing software at one time. You cannot overwrite running software, so OTA firmware is loaded in the opposite partition and the bootloader loads that on next boot.

tuya-convert works by first presenting the Tuya firmware with our [intermediate firmware](https://github.com/vtrust-de/esp8266-ota-flash-convert). This is the stub which then connects back to your machine, allowing you to create a backup, modify the bootloader, and upload custom firmware. It also takes care of normalizing the flash space so that when you get your firmware loaded, it is in the correct place and doesn't crash.

The backup created is of the entire flash memory, we don't know which partition has the original firmware, and there are several other important partitions needed by the Tuya firmware. It must be placed back exactly as it was; it will not run if it is in the wrong place as memory addresses are absolute. Since we can't overwrite the firmware that is running, there is no way to load the entire flash space OTA.

closed 2020-07-20 by kueblc

tonyfitzs, 2020-07-20:
Thank you Colin.

This is most helpful. Its clear to me now