ct-Open-Source/tuya-convert #708: Arlec Smart bulb flashing issues.

jonaseymour, 2020-07-20:
I've just tried to flash both the [Arlec GLD124HA](https://templates.blakadder.com/arlec_GLD124HA.html) and [GLD060HA](https://templates.blakadder.com/arlec_GLD060HA.html), and having no luck. Just keeps 'Resending SmartConfig Packets'

This seems to be the key issue in the psk logs if it helps (repeats quite a few times):
`could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)`

Does this imply that the device is not compatible with tuya-convert?

Also flashed a few [Mirabella Genio 1002334](https://templates.blakadder.com/mirabella_genio_1002334.html) with no such issues, so I'm sure I'm doing the process correctly.

Any help appreciated.

kueblc, 2020-07-20:
If you see a PSK ID starting with `02`, you're correct, it is unfortunately not compatible with tuya-convert at this time (#483)

jonaseymour, 2020-07-20:
> If you see a PSK ID starting with `02`, you're correct,

Yep `ID: 0242416f68626d643661473931494652316771f562f9eec20bda9fc26066cd6bab8e5ee1262736d88fb77fa84951a99578
`
Bugger

> it is unfortunately not compatible with tuya-convert at this time (#483)

Does that mean _maybe_ in the future?

kueblc, 2020-07-20:
Maybe, sure, but it will take a new exploit to be discovered and purposed, so there is no guarantee if or when.

jonaseymour, 2020-07-20:
Thank mate. Appreciate the work on this regardless!

kueblc, 2020-07-20:
Glad to help. Keeping my fingers crossed we find another workaround.

closed 2020-07-20 by kueblc