ct-Open-Source/tuya-convert #709: Stuck on 'Attempting to flash tasmota.bin, this may take a few seconds...'

Draxter, 2020-07-20:
The process is getting stuck after chosing the bin to flash with, I have given it about 15 minutes and eventually had to exit with ctrl+c. The device in question is this: https://templates.blakadder.com/julun_JL-021.html

After rebooting the device it comes back with a vtrust-recovery SSID.

Terminal Output:

Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
...................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  63879      0  0:00:16  0:00:16 --:--:-- 28970
curl: Saved to filename 'firmware-4be7a.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 4be7a
MAC: D8:F1:5B:04:BE:7A
BootVersion: 7
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 144068
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v8.1.0.2 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...

LOGS:

smarthack-psk.log:
new client on port 443 from 10.42.42.39:51552
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.39:51553
new client on port 443 from 10.42.42.39:51557
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.10:142
ID: 0142416f68626d64366147393149465231ebf4aa7a36fe5b93a3557fa68e4b1a845fcaeb57ecf83a187b59927ffeccb1f30d
PSK: d53840ac1fd766145f29a60a74d7c68ffc75fbfe4f02c45a0133ddd229305dcf
new client on port 443 from 10.42.42.10:30664
[... 184 lines cut ...]
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr 72:8c:e9:8b:04:34 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 50:bc:96:67:63:49
wlan0: AP-STA-CONNECTED d8:f1:5b:04:be:7a
wlan0: AP-STA-DISCONNECTED d8:f1:5b:04:be:7a
wlan0: AP-STA-CONNECTED d8:f1:5b:04:be:7a
wlan0: AP-STA-DISCONNECTED 50:bc:96:67:63:49
wlan0: AP-STA-CONNECTED 50:bc:96:67:63:49
wlan0: AP-STA-DISCONNECTED d8:f1:5b:04:be:7a
wlan0: AP-STA-CONNECTED d8:f1:5b:04:be:7a
wlan0: AP-STA-DISCONNECTED d8:f1:5b:04:be:7a
wlan0: AP-STA-CONNECTED d8:f1:5b:04:be:7a
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED d8:f1:5b:04:be:7a
wlan0: AP-STA-DISCONNECTED 50:bc:96:67:63:49
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...

kueblc, 2020-07-20:
Can you reboot the device with `start_flash` still running? You can skip attaching a phone to `vtrust-flash`.

Draxter, 2020-07-20:
> Can you reboot the device with `start_flash` still running? You can skip attaching a phone to `vtrust-flash`.

Certainly can. After some more messing around I am not even getting the 'vtrust-recovery' SSIDs from the two lightbulbs that I tried, so I guess I bricked them.
I still have one more stock one that I haven't touched. Could you outline the entire process to me? Is rebooting the device that is in recovery mode while start_flash is still running supposed to help recovert it back to stock firmware?

Draxter, 2020-07-21:
I can returrn to stock firmware fine, thanks for the /undo trick, didn't realise that was a thing. But still no luck flashing Tasmota at all. Perhaps they patched the firmware. Tried all combinations of rebooting the device at various stages of tuya-convert running and no joy, still stuck on 'Attempting to flash tasmota...' screen.

closed 2023-08-25 by Draxter