ct-Open-Source/tuya-convert #711: New, not flashable bulb, help?

spcqike, 2020-07-22:
hi all,

i recently bought a pair of smart light bulbs (LOHAS E14 from Amazon https://www.amazon.de/dp/B0796MX6DP/), that seems to be not compatible.
flashing over wifi with tuya convert fails, like many others, because of new firmware/ssl?

after i opened one bulb i found out, that it's kind of tricky, if not impossible (for me), to solder and flash it. the TX/RX/GPIO0 pads are hidden on the lower side of the PCB.

if anyone has an idea, where to measure or solder, i can try. but i guess, it's not compatible. :(

best regards

[image]
[image]
[image]

JWSmythe, 2020-07-24:
Do a Google search for "TYWE2S pinout".  There are a bunch of little modules like you have, that are used in these lights, and they all have similar names.   

https://docs.tuya.com/en/iot/device-development/module/wifi-module/wifie2smodule?id=K9605u79tgxug

You'll also find a lot of these modules by searching for the FCC page. 

https://fccid.io/2ANDL-TYWE2S/Users-Manual/Users-Manual-3596121

Sometimes they keep the pinout restricted.  I have a couple lights here with the TYWE2L chip, and they didn't have the manual available for the first year.  That just expired a few days ago.  Whoohoo!  

For yours, viewing where there are 6 silver and 5 copper pins, the pins you're interested in are...
1 - 3.3V
3 - GND 
5 - RX 
7 - TX

There are just a few GPIO pins on these.  It'll save you a lot of guessing.  They are 4, 5, 12, 13, 14.

If this is like some I've played with, just start setting each of those to be PWM, and see what lights up.    

If that doesn't work, you'll need to use your multimeter, and see what is connected to the GPIO pins with no resistance, and then check the specs on the chip they are attached to.

Good luck.

JoelMSawyer, 2020-07-27:
I managed to flash a couple of LOHAS GU10 bulbs with tuya-convert, and they have very similar internals based on your photos. In case it's useful, here is my esphome config for the GPIO:

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO5
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO4
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO13
  - platform: esp8266_pwm
    id: output_warm_white
    pin: GPIO12
  - platform: esp8266_pwm
    id: output_cold_white
    pin: GPIO14

I hope this is useful, if you haven't got it working already.

spcqike, 2020-07-28:
thank you both, but i meant, that it's not possible to remove the blue module (TYWE2S) from the white PCB. it's soldered flat (reflow?) and i dare not solder it. i would need a hot air gun / hot air solder station and heat the hole module to ~ 300°C. i'm afraid that i damage something...

so, this post was more a "warning" to others, who might buy these modules :) as long as tuya convert doesn't work with the new firmware, it's not possible or at least really difficult to flash theses bulbs.

JoelMSawyer, 2020-07-28:
That's a shame - I also have a couple of LOHAS bulbs's which were unflashable OTA due to the new firmware, and I was going to look into flashing the TYWE2S - but as you say it appears like it would require additional equipment.

JWSmythe, 2020-07-28:
You should be able to solder onto the fingers sticking down from the module.  The part with 6 silver and 5 gold parts.  It may be necessary to scrape some paint.  It may be necessary to drill through just the module's PCB (not the PCB holding the LEDs), but only on 1,3,5,7.   

If you look up the other ICs on the opposite side, you can probably find the one responsible for handing the low voltage power over to it.  Clip on those for 3v3 and GND with your TTL adapter.  With that, you'd only need connect to pins 5, and 7 on the module.    

You can accomplish all that with a regular soldering iron, TTL adapter, razor knife, and maybe a dremel.

Don't give up.  You already voided the warranty.  You may as well make it work.  You're almost there.

spcqike, 2020-07-28:
Thanks for your suggestion, but I still need GPIO0 BC it must be hold LOW for flashing :) or at least to enter flash mode, mustn’t it?

We already know the design of the TYWE2S, so we know that the “silver” pins are 3V3, GND, TX and RX. If I manage to connect to TX and RX, it should be easy to get 3V3 and GND too. But what about GPIO0? :( 

Somehow I can’t imaging that it’s used to control one of the PWMs. And I can’t find any information about the LED driver. Just that it uses a PWM from the esp. my guess is that 5 PWMs (one per color and one per white (warm and cold)) are used. And as we have 7 possible GPIOs on the lower side of the module, It would be to easy if GPIO0 would have been used to control one PWM .... 

Maybe it’s best to wait till WiFi flash is possible again :)

spcqike, 2020-08-03:
Just a quick update:

A friend of mine managed to desolder the esp-Module from The LED PCB. 

If anyone is interested, here are two pictures. 
As thought, GPIO0 isn’t used to control a pwm. So one must desolder the PCBs or the metal shield from the esp directly. 

[image]
[image]

JoelMSawyer, 2020-08-03:
Thanks for a useful update - gives me some confidence that I could attempt this. Was it a particularly difficult task to de-solder, just a conventional soldering iron used?

spcqike, 2020-08-04:
Hey,

I've asked my friend. They've used a [RD-500V rework machine](http://denondic.co.jp/en/doc_rd500v.php). So nothing one has at home. but he said, it's possible to do similar things with an hot air rework station. it may need some training/testing, so nothing that you would like to test on your first bulb :) 
a conventional soldering iron is not sufficient.

JoelMSawyer, 2020-08-04:
Definitely not then! - thanks very much for enquiring - will wait until tuya-convert hopefully is able to flash bulbs with the newer firmware.

jdubz666, 2020-08-26:
Hi thought I would ask this here if that is all good. 

Any idea if these A19 Lohas light bulbs can be flashed over wifi or will they run into the same issue as the E14 bulbs?

https://www.amazon.com/LED-LOHAS-2700K-6000K-Daylight-Equivalent/dp/B07T7W7ZMW

Thanks

chrisphotonic, 2020-08-27:
I just bought the A19 Lohas in the link above and I can't flash with Tuya convert either of course. Here are the internal pics:

[image]

[image]

[image]

I haven't started probing for voltages yet but I'm about to if no one else has done this yet.

jdubz666, 2020-08-27:
> I just bought the A19 Lohas in the link above and I can't flash with Tuya convert either of course. Here are the internal pics:
> 
> [image]
> 
> [image]
> 
> [image]
> 
> I haven't started probing for voltages yet but I'm about to if no one else has done this yet.

Darn... Thanks for the update

jdubz666, 2020-09-13:
Hey @chrisphotonic did you probe for voltages with the Lohas A19. Do you think it will be possible to create a template for these?
Thanks.

jerrit, 2020-12-18:
@chrisphotonic / @jdubz666 Any luck with manually flashing these? 3/4 worked with tuya convert but I have one I need to manually flash...

technowizard79, 2021-03-30:
> As thought, GPIO0 isn’t used to control a pwm. So one must desolder the PCBs or the metal shield from the esp directly.

Actually the cover pries off fairly easy.  I just used a scribe on mine, though on another one I did have to use a small pair of needle nose to wiggle one side completely loose.

[image]

[image]

technowizard79, 2021-04-08:
For the original Lohas Candelabra bulb question, I can confirm that with some moderate soldering ability you can flash the ESP chip **without** desoldering the TYWE2S module from the LED board.  I'll add pictures later when I flash another one, but in the meantime here's the instructions:

- First you need to do like I mentioned in my previous post, and remove the TYWE2S module cover.  Again it pries off fairly easily, and so far I found the top of the module (wifi antenna side) has the best gap to start prying from.  Don't be afraid to give it a little effort, though stay mindful of the sensitive components that are present underneath it.  
- Next you need to remove the white-ish silver enamel / solder mask between top left of GPIO13 to top right of GPIO05 front copper pads.  This will expose the (albeit tiny) pin points for the TX, RX, Ground, & 3.3v on the front of the TYWE2S module.  They're directly inline with the neighboring visible pin points on the copper pads.  If you have the LED board completely removed from the candelabra light base, then you only need to uncover the middle two located to the left and right of the GPIO13 pad.  As @JWSmythe pointed out above, pin 6 on 1F8JK (this designation varies slightly from bulb to bulb) / MP2359DJ-LF on the back of the LED board is the 3.3v.  You don't have to use pin 2 for ground though, as you can just use the negative lead that normally powers the board (i.e. the white wire / pad that connects the LED board to the light's socket base).
- Now you need some fine single strand wire to carefully solder to these newly exposed points.  It can be a bit daunting if you're not used to soldering things this small, but it's a lot easier and safer to solder here than it is to the ESP chip directly.  Be sure to check with a DMM that you're newly soldered wires are not inadvertently shorting to the neighboring copper pads.  Afterwards you'll need to attach the fine wire strands to your normal serial flashing wires.  We're not going to the Moon with this thing, so a light quick and dirty solder job is fine... you don't exactly want to be tugging or overheating your fragile wire strands.
- Lastly you'll need a fine test lead to momentarily tough and ground out GPIO0 (pin 15 on the ESP chip as shown in my previous post) while you plug in your USB serial flasher.  If you don't have one, you can fashion one from a sewing needle, stranded wire, and some electrical tape.  Remember to be gentle when touching the ESP pins, as they're just soft solder pin legs, and they will deform and get damaged if you press too hard.

That's pretty much it.  When you're done flashing desolder your fine wire strands and reassemble the light.  For those that are curious, there is no need (nor should you really try) to reattach the TYWE2S module cover.  It's just there to protect the components during manufacturing and distribution.  Also I recommend a magnifying visor, glass, or scope to accomplish all of this... the pins and soldering tasks are pretty small, and you don't want to short or touch the wrong things while working on this.  Visors go for as little $15 on Amazon, and they make this project a heck of a lot easier!

ferbulous, 2022-06-02:
Hi, sorry to bump this year old issue 
I just have to ask, for anyone who did desolder the chip using hot air gun similar to led bulb posted by @spcqike 

How difficult can it be without melting/damaging leds surrounding the chip?

technowizard79, 2022-06-03:
FYI he didn't use a "hot air gun," but rather a professional level solder rework station.  Industrial/commercial ones like he listed can easily cost thousands of dollars, and sometimes tens of thousands!  Using a heat gun or entry level consumer rework set is pretty much guaranteed to damage the components, especially if you're inexperienced at it.  Where this one is more difficult than say the Costco dimmer switches that also uses this module that some desolder, is that it has a few pads soldered down on the back of the board.  You're just super likely to damage _something_ while solder flowing it off **AND** then back on.

The method I listed above is doable by even novice electronic techs, and only requires a decent soldering iron, a magnifying glass/headset/scope, some fine wire, and a very steady hand.  I've done nearly a dozen of these bulbs this way, and haven't lost one yet (though came close once while removing the ESP chip cover... broke a smd cap off, but was able to find/solder it back on).  If you remove the board from the bulb housing (which you would have to do if you planned to desolder the entire TYWE2S module as you alluded to), then you only need to uncover the TX and RX points, as you can use the voltage regulator IC pin 6 (3.3v) and the white negative power lead (ground) as listed above.

Obviously feel free to wait for someone else to chime in here if you'd like, but unless you have the equipment and skills to do this professionally, then my professional opinion is you're going to absolutely damage the components.

JWSmythe, 2022-06-03:
Hot air rework stations aren't expensive.  Please don't try to talk with authority about topics you know nothing about.

It is perfectly practical for him to desolder and resolder any component shown in the pictures.   I can.   It only cost me about $137, and I use it all the time.

This is the hot air rework station. It was only $90 USD.  I could change every single smd component in the light with no problems.  There are a *bunch* of these, sold under different brand names.  They are all the same thing, from the same factory.    Don't try to compare reviews between them, one that looks the same isn't any better or worse than another.  It's just the randomness of who reviewed on which.  Buy the cheapest one you can find. 

https://www.amazon.com/gp/product/B07SCPZJYS

My only problem is when I start working with the really small components, like the resistors around the ESP8285 chip in the photos.   So I use this microscope.  It cost $47 USD.  The same thing about different brands selling the same one.  They're all the same.  Buy the cheapest one you can find.

https://www.amazon.com/gp/product/B081L4Y98G

There is a flaw with that microscope.  For some reason, the board stays on after it's shut off, and won't let it turn back on until the battery dies.  If you get it, pop the top off, and disconnect the battery.  You don't need it if you use it with USB power.  I either plug it into my computer, and use the Windows Camera App to view, or I pull it up on my phone or tablet.    You can clearly see the smallest traces, so it's more than you'll need.   It isn't like a biology microscope, so you won't see cells and microbes. :) 

There are other small tools and consumables that you will need.   

- a flux pen (I use this the most)
- flux paste and/or flux liquid in a syringe.  The one for electronics, not plumbing.
- solder (wire).  For electronics, not plumbing.
- solder paste.  
- solder sucker (the blue one, it's cheapest)
- desoldering wick (braided fluxed copper in a spool)
- tweezers (like, the sharp beauty tweezers, and the blunt angled beauty tweezers, from Walmart or whatever)
- pick (like a dental pick)
- electronics repair toolset, with all the bits.   You don't need it for this project, but you will for others.  Walmart stores stock a decent starter kit for $15.  I have a bigger kit that cost closer to $100, with all kinds of obscure bits that you'll need once ever. 
- 91% or higher isopropyl alcohol, for cleaning boards, and cleaning up flux after soldering.  Health/beauty kind.
- q-tips for cleaning, the beauty/ear  kind.
- head mounted magnifier.  I use this a lot, if the work is small, but not small enough to require the microscope.

On the ones I noticed not as plumbing, they use a different flux.  Their flux will make a mess of your electronics, and it will corrode pretty soon after you've done the work. 

And, finally, don't bump unrelated posts to ask new questions.  You could have asked this on reddit or somewhere, and gotten more useful answers.   I just didn't want you thinking technowizard was right, and this is an impossible as a hobby.  I bought some of that to fix a blown diode on an ESP8266 dev board.   I bought the rest to expand my ability.  Now I will fix most any electronics that I come across, and I have been designing new boards, and having them manufactured.   

At some point, I will be making a replacement module for lights like this, to replace the Ali/Tuya module that we can't flash.  I want to make those with a USB port on them too, but it might just get marked header pins.  It'll depend on how small I can make it.  I have designed some really small electronics, and that isn't beyond my ability, I just haven't gotten around to it.

technowizard79, 2022-06-03:
> Hot air rework stations aren't expensive. Please don't try to talk with authority about topics you know nothing about.

You're right, I've only been working with this stuff on multi-million dollar electronics for 30 years... what do I know?

Seriously though, I didn't say "hot air rework stations" were expensive... I said the one that **spcqike** used (a Den-On RD-500V unit) was an industrial grade one, and those type are in fact quite expensive (at least for a regular consumer/hobbyist).  Also I didn't say it was "impossible" to accomplish the removal of the module with one of the sub-$100 cheapy hot air stations... no I just determined that the fact the poster was even asking the question, let alone calling the RD-500V a "hot air gun," that he was likely very inexperienced with this type of work, and that he was "pretty much guaranteed" and "super likely" to melt/damage components.

@JWSmythe I'm glad to hear that people like yourself are getting proficient with more advance tools, and passing that knowledge on to others.  That said, don't presume I know nothing about this topic just because I cautioned an inexperienced electronics person (who let's be honest is probably just wanting to up their smart home game, and nothing more) towards an alternative and novice friendlier method.  As I said above I do this professionally for a living, and I made and posted the above directions for this device and others after reading/seeing numerous ESP based module boards and various components getting destroyed by cheap Amazon hot air setups, and overall lack of skill/experience by smart home hobbyists.

technowizard79, 2022-06-04:
>Hot air rework stations aren't expensive. Please don't try to talk with authority about topics you know nothing about.

You're right, I've only been working with this stuff on multi-million dollar electronics for nearly 30 years... what do I know?

Seriously though, I didn't say "hot air rework stations" were expensive... I said the one that **spcqike** used (a Den-On RD-500V unit) was an industrial grade one, and those type are in fact quite expensive (at least for a regular consumer/hobbyist). Also I didn't say it was "impossible" to accomplish the removal of the module with one of the sub-$100 cheapy hot air stations... no I just determined that the fact the poster was even asking the question, let alone calling the RD-500V a "hot air gun," that he was likely very inexperienced with this type of work, and that he was "pretty much guaranteed" and "super likely" to melt/damage components.

@JWSmythe I'm glad to hear that people like yourself are getting proficient with more advance tools, and passing that knowledge on to others. That said, don't presume I know nothing about this topic just because I cautioned an inexperienced electronics person (who let's be honest is probably just wanting to up their smart home game, and nothing more) towards an alternative and novice friendlier method. As I said above I do this professionally for a living, and I made and posted the above directions for this device and others after reading/seeing numerous ESP based module boards and various components getting destroyed by cheap Amazon hot air setups, and overall lack of skill/experience by smart home hobbyists.

ferbulous, 2022-06-06:
Thanks guys, for clearing that up
I don’t have a hot air workstation yet
I’ve been using hot air gun to desolder beken chips and replace with esp chip to flash tasmota.
It’s been quite an easy task especially when the beken chip is far away from any fragile components that might get damaged from high temperature.

But recently, i have 2-3 switches with the beken chip sitting right next to the button switch.

If i were to desolder this and not damage/melt the button switch next to it. 
What would be the best approach with hot air rework station?

[image]
[image]