ct-Open-Source/tuya-convert #714: install_prereq.sh error

JBB29, 2020-07-24:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

JBB29, 2020-07-24:
Install script exits with error 128 in none and I have no clue where to start. 
The last lines of the terminal output reads: 

Cloning https://github.com/drbild/sslpsk.git to /tmp/pip-req-build-s3550pjx
fatal: unable to access 'https://github.com/drbild/sslpsk.git/': Could not resolve host: github.com
Command "git clone -q https://github.com/drbild/sslpsk.git /tmp/pip-req-build-s3550pjx" failed with error code 128 in None

Any ideas? 
Thanks, John.

kueblc, 2020-07-24:
> Could not resolve host: github.com

That's a temporary network error. Clearly you are here now, so try running it again.

JBB29, 2020-07-24:
I did see the "Could not resolve host: github.com" but could still ssh in, and it was late. Guess I needed confirmation. 
Tried to ping and saw that could not resolve any address but could ping an IP. Reboot fixed the ping and allowed the tuya script to finish. Guess is that the install of dnsmasp upset something but seems to be working now. 
On to flashing. 
Thanks for your help.

kueblc, 2020-07-25:
No problem

closed 2020-07-25 by kueblc