ct-Open-Source/tuya-convert #716: Help finding OTA flashable E26/E27 bulb base?

hartzell, 2020-07-26:
This feels like a stretch for a GitHub issue, but I haven't found a better place to ask.  I'd be happy to be redirected.

I'm trying to find an E26/E27 bulb base that's suitable for incandescent bulbs that I can OTA switch to Tasmota.  I've tried the [AIcase](https://www.amazon.com/gp/product/B07BTQJG6J) (which had reports of successful conversion) and the [wrpacttg](https://www.amazon.com/gp/product/B0834VWJGH/) units.  Both sets that I received appear to have been updated to the newer firmware and suffer from #483.

I did some minor disassembly on the AIcase and I don't see a good way to get access to the pins to do a hardwired upgrade.

I've ordered a Sonoff Slampher, which is not OTA upgradable but has an easily accessible set of pin headers.  The base might turn out to be too big for my use case though, so I'm still looking.

I'd be happy to find someone that still has "New Old Stock" of the AIcase or wrpacttg bases....

hartzell, 2020-08-02:
FWIW, I received the Slampher.  It seems to be a newer version of version 2 that can be easily programmed using just the four pin header; it goes into programming mode by holding down the button, unlike the previous versions which required grounding `R9`.

seritos, 2020-09-29:
@hartzell May I ask you where have you seen any succeed AIcase bulb conversion? 

Thnx!

hartzell, 2020-09-29:
e.g. Will Hutcheon's comment here: https://www.amazon.ca/product-reviews/B07JGC1FYG?ref_=fsclp_pl_nr_11

seritos, 2020-09-29:
Thanx, @hartzell . Just purchased a pack of 2 AIcase and see what I can do. 
As far as I could read, there is no way to do it trough Tuya-convert... isn't it? May I ask you if the unit has to be scratched to be disassembled? I care about this

...did you tried to disassembly the wrpacttg? 

Thnx

hartzell, 2020-09-29:
I bought a handful of different ones, disassembled something (might have been the AICase) and ended up cracking it.  Then I wasn't able to find a reasonable way into the flashing pings.  Punted, returned the units that I hadn't broken and went with the slamper.  It just works.

seritos, 2021-01-26:
Anyone has found a flashable good ones?