ct-Open-Source/tuya-convert #717: Trying to flash Fiet SYSW-BR30RGBAGT2 -- Doesn't get to intermediate stage

cmoski, 2020-07-27:
Hey all -- 

Just picked up some of these bulbs at Costco. They are the SYW-BR30RGBAGT2 / Feit BPBR30/RGBw/AG/2(C) with date code 17-20-19S 11

I am able to get the bulb to connect to the wifi network, but it doesn't seem to to take the intermediate flash. Web doesn't report any traffic, psk doesn't report anything either, just see the device connected in the wifi log.

Any assistance would be appreciated. Took one of them apart and the actual module is the WB3L

(https://docs.tuya.com/en/iot/device-development/module/wifibt-dual-mode-module/wb3l-module-datasheet?id=K9duiggw2v8sp)

sacdroid, 2020-08-17:
@cmoski were you able to get anywhere with flashing these feit br30 bulbs?

cmoski, 2020-08-28:
@sacdroid Hey! Sorry I didn't see your reply until now. 

I gave up on flashing these bulbs and return them for a different model, sadly. I still have one if you have any ideas though :)

bani6809, 2020-11-10:
Looks related to #814

kueblc, 2020-11-10:
@bani6809 is probably right. Closing as the issue author has come to their own resolution.

closed 2020-11-10 by kueblc