ct-Open-Source/tuya-convert #718: Please edit WLAN in config.txt to one of: enp0s3, lo, wlx08beac0aaa9d

ZivAssor, 2020-07-28:
Hi,
I am using Ubuntu 20.04 VM (VirtualBox) running on macOS.
I have EDIMAX EW-7811Un WiFi adapter.

When executing `./start_flash.sh` I get this error:

> Checking for network interface wlan0... Not found!
> Please edit WLAN in config.txt to one of: enp0s3, lo, wlx08beac0aaa9d
> Cleaning up...
> No screen session found.
> No screen session found.
> No screen session found.
> No screen session found.
> Closing AP
> Exiting...

Any ideas what to do?

spcqike, 2020-07-28:
Hi,

as seen in your error message, you either don't have a wifi adapter, or it's not called "wlan0". you can check with "ipconfig", what it's called. than you need to change the name in config.txt.

ZivAssor, 2020-07-28:
Worked! Thanks

closed 2020-07-28 by ZivAssor