ct-Open-Source/tuya-convert #720: Prereq install failure with Ubuntu Flash Drive

atdemeo, 2020-07-30:
Hello,

I'm attempting to install the Tuya-Convert on an UBUNTU live Flash drive.  I have attempted this on 20.04 LTS and 18.04.  Same error after the following:
```
# sudo Apt Update
# sudo Apt Upgrade
(Everything up to date)
# git clone https://github.com/ct-Open-Source/tuya-convert
# cd tuya-convert
# sudo ./install_prereq.sh
```
  And the resulting code:
```
ubuntu@ubuntu:~/tuya-convert$ sudo ./install_prereq.sh
Ign:1 cdrom://Ubuntu 18.04.3 LTS _Bionic Beaver_ - Release amd64 (20190805) bionic InRelease
Hit:2 cdrom://Ubuntu 18.04.3 LTS _Bionic Beaver_ - Release amd64 (20190805) bionic Release
Hit:3 http://archive.ubuntu.com/ubuntu bionic InRelease                        
Hit:4 http://archive.ubuntu.com/ubuntu bionic-updates InRelease                
Hit:5 http://security.ubuntu.com/ubuntu bionic-security InRelease              
Reading package lists... Done                      
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package dnsmasq is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  dnsmasq-base

Package haveged is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python3-wheel is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'dnsmasq' has no installation candidate
E: Unable to locate package hostapd
E: Unable to locate package python3-pip
E: Package 'python3-wheel' has no installation candidate
E: Unable to locate package mosquitto
E: Package 'haveged' has no installation candidate
```
I've tried new drives, switched to and from Persistence Storage for both version attempts, haven't been able to get past this point.     I'm brand new to Linux, so I imagine I'm missing something. 

I've ordered a PI to give it a shot on there, but This should still work....right?  Thanks for your help

kueblc, 2020-07-30:
> sudo Apt Update

Linux is case sensitive, which you might not be used to if you're coming from Windows. They are mostly lowercase.

`dnsmasq`, `haveged`, etc, are all present in the current Ubuntu repositories. Running from a live disk is not supported, so if your live disk is having trouble locating these packages, you may need to start the OS normally.

atdemeo, 2020-07-31:
Live disk was the issue. I thought with persistence it would work as I found that others had success with that, but definitely wasn't working.  I achieved success with Ubuntu install LTS 20.04 partitioned alongside windows 10.  

Thanks for the fix.

Closing this thread and moving on to Intermediate Firmware errors.

closed 2020-07-31 by atdemeo