ct-Open-Source/tuya-convert #721: WiFI-W6B

DJ-Q, 2020-07-30:
I have a "No Neutral" 2 gang touch light switch.  The markings on the board say WIFI-W6B and i can see a ESP8285 on board.

I can't get this to flash using tuya convert.  The switch seems to connect to the AP and then immediatley comes out of pairing mode.  Tuya convert just says 

SmartConfig complete.
Resending SmartConfig Packets

over and over and then fails with 

Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n

I have used the same setup to successfully flash smart plugs - so know my installation and method works - is there anything that would stop this switch from accepting tuya convert?

kueblc, 2020-07-30:
> Check the *.log files in the scripts folder

Can't help much without these

> is there anything that would stop this switch from accepting tuya convert

New firmware (#483)

broekhuijsen, 2020-08-07:
Hi there,

I have exactly the same device. I have tried many many times, but it does not work.

After failing all these times, I have installed the tuya app to see if it even would work. And it does.

Checked the firmware

WiFi 1.1.1
MCU 1.1.1

I hereby upload my files, hoping it is helpful.

[](url)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5044279/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5044280/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5044281/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5044282/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5044283/smarthack-mqtt.log)

broekhuijsen, 2020-08-08:
Here some pictures of the board.

It also supports 433 Mhz

[image]
[image]

duvalt, 2020-08-08:
Hi,
Same model for me and same problem ...
Do you know where to connect wires (3.3, GND, TX, RX and GPIO-0) for wired flashing ?
Thanks in advance.

broekhuijsen, 2020-08-08:
> Update
> Find this : https://codesandbolts.com/girier-smart-wall-switch-wifi-w6-b-custom-firmware/

Wow this is a great find. I have been looking on the internet for this, but could not find anything and was going to try this today with trial and error.

Now I can use my saturday for other fun things :D and there are still two of these in the mail, so I was already afraid to be stuck with tuya for these.

Thanks.

broekhuijsen, 2020-08-08:
@duvalt can you mange to flash the device with this pin out? For mine, it doesn't seem to work with my board :(

duvalt, 2020-08-08:
> @duvalt can you manage to flash the device with this pin out? For mine, it doesn't seem to work with my board :(

Hi, not yet, waiting for ftdi/usb converter, should be delivered in 10 days ...

broekhuijsen, 2020-08-08:
> > @duvalt can you manage to flash the device with this pin out? For mine, it doesn't seem to work with my board :(
> 
> Hi, not yet, waiting for ftdi/usb converter, should be delivered in 10 days ...

Ok, well I was not able to do so. I have flashed many devices already with tasmotizer, but this one just won't budge. Now the gpoio0 came lose from the pcb so it is game over for this one. :'( 

Maybe there will be a new version of tuyaconvert in the future that will work. I hope I will have more luck with the other two that are coming in. 

I wish you good luck, let me know if you managed.

joeldavidw, 2020-08-11:
Hi! I am the owner of the blog mentioned above. Sorry to hear that you were not able to flash your device! Not sure if this information would help but I used this [CP2102 USB to TTL UART from AliExpress](https://www.aliexpress.com/item/32717057832.html) to flash my device.

broekhuijsen, 2020-08-11:
> Hi! I am the owner of the blog mentioned above. Sorry to hear that you were not able to flash your device! Not sure if this information would help but I used this [CP2102 USB to TTL UART from AliExpress](https://www.aliexpress.com/item/32717057832.html) to flash my device.

Thanks for the tip. I however have read that there is not really a huge difference between what I am using and what you used:

https://www.banggood.com/FT232RL-FTDI-USB-To-TTL-Serial-Converter-Adapter-Module-p-917226.html?rmmds=myorder&cur_warehouse=CN

So far I have been successful at flashing almost everything. I gues this board just didn't want to cooperate :D

Soon I will receive the other switches at home, let's see if I  will be able to flash those. If not, my plan is to hide sonoff basics in my ceiling and then to trigger them with RF buttons through the Sonoff RF bridge and my home-assistant.

https://www.aliexpress.com/item/4000415317711.html?spm=a2g0s.9042311.0.0.3f6c4c4dBkmWJ3

adamb94, 2020-08-13:
Hi, to flash this you need to Exchange RX with TX. It was successful for me.

Do you have pinout for button and relays for this?

iondulgheru, 2020-08-13:
I somehow wrote the firmware using serial with the reset pin also connected to ground while booting, and removing the connection from ground after i run esptool command. 
But I also don't have the pinout for button and relays for this.

broekhuijsen, 2020-08-13:
I tried exchanging the RX and TX pins, but no success. I also tried flashing with GPiO0 constantly connected, but nope.

I still didn't receive my other switches, because, well Portuguese postal service... sigh...

Let's see if I manage next time. It would be cool if one of those chinese companies would release all their devices with a minimal web page to flash your software. Or even better, release the hardware with tasmota pre-installed.

There should be a market for that.

broekhuijsen, 2020-08-13:
Just one thing, did you manage to flash it with the module mounted in the switch? Because I took it out to flash it.

adamb94, 2020-08-13:
I disconnected W6B-WiFi from rest of switch and had GPIO0 connected to groud constantly during flash. I used cheapest Uart cable found on aliexpress :-) You need to have flash for ESP8285 1MB

DJ-Q, 2020-08-13:
just to confirm- i have successfully flashed tasmota onto this switch using USB-UART adapter and following the pin out above, swapping RX and TX

Thanks to the blog author for providing the info.

joeldavidw, 2020-08-14:
@iondulgheru @adamb94 I used this template https://templates.blakadder.com/zuczug_3gang.html for the buttons and relays.

Schwiings, 2020-08-15:
Hi
Thanks to @joeldavidw blog I was able to flash this device but somehow the tactile button doesn't work anymore. I am still able to toggle the switch via WiFi but not physically.
Did you experience the same issue?

broekhuijsen, 2020-08-17:
Hmm, I have received the new boards, and again I cannot flash. But I might have found out why.

I have bought a fake FTDI chip apparently. Gonna order a real one in a store nearby.

broekhuijsen, 2020-08-19:
Yesss success!!! The FTDI device that I received, turned out to be a fake as well, and it didn't work. But I also ordered a device based on a PL2303 chip (the only other one they had at that store). This chip officially does not work with Windows 10, but [luckily someone found out](http://wp.brodzinski.net/hardware/fake-pl2303-how-to-install/) that installing a very old driver resolves that problem.

I even managed to flash the first switch that lost it GPIO0 pad by carefully scratching some of the layer on top of the copper track that lead to that pad.

broekhuijsen, 2020-08-20:
I have submitted a template for this device, but it is not online yet.

`{"NAME":"JR-SEU01","GPIO":[255,157,255,255,255,18,0,0,21,255,255,255,255],"FLAG":7,"BASE":2}`

moshe19803, 2020-10-03:
What software you use  to flash?

broekhuijsen, 2020-10-03:
> What software you use to flash?

I use tasmotizer, very easy to use.

I never managed to flash with an FTDI, even not a real one.

moshe19803, 2020-10-03:
So what hardware you use if you don't use FTDI?
tasmotizer with?

broekhuijsen, 2020-10-03:
The PL2303, but you must install an [old driver.](
http://wp.brodzinski.net/hardware/fake-pl2303-how-to-install/)

juan11perez, 2020-10-04:
@joeldavidw  thank you for posting this pinout. It worked for me as you've laid it out.

However, I could not flash with tasmotizer. It would connect and brake at about 14% progress.
I was able to flash with Pyflasher and these settings:
boud rate 921600
flash mode DOUT

@broekhuijsen  thank you for the template. this switch also provides power consumption. Great!!

honzapatCZ, 2020-12-21:
[image]
Just pull RST to GND and no need to cut traces or whatever
General rule for unmarked chips is 1 or 4 is RST try to pull it UP or DOWN and it will prob. work

kwirk, 2020-12-24:
Some really useful information in this thread. Thanks all for sharing.

So I bought two similar switches, a 1-gang and a 2-gang version. They are slightly different from the one in the picture, as mine are missing the 433MHz components, missing the buzzer (despite the instructions saying it should buzz/beep), and has an blue LED above flashing pins.

The 1-gang flashed (with ESPHome) no trouble whatsoever, using pinouts from @joeldavidw's blog, but the 2-gang was being stubborn, producing timeout and invalid header errors from *esptool*. I tried re-soldering, checking continuity back to esp8285 pins, using an external power supply, …

In the end I was able to flash the 2-gang switch by using a sequence of: powering up the device **without** GPIO0 low; putting GPIO0 low then briefly putting RST to low (that's the unlabelled pin between GPIO0 and GROUND from @joeldavidw's blog) to reset the esp8285; and then finally *esptool* worked.

9magic, 2021-01-14:
Hi,

@honzapatCZ thank you for the info about RST pin and a nice picture.
I have two unbranded WiFI-W6B switches. I could not use tuya-convert any more. I needed to connect this tuya mcu RST pin to ground and then I flashed with tasmotizer successfully.

I set up the switch without neutral line in the wall and connected capacitor in parallel to the lightbulb.
I used @broekhuijsen's template from above but the power readings were swinging around 400W when the light was out and did not change significally when I turned it on.

I cut the power to the switch and went to bed:)
Next day I tested actual consumption from the other switch and concluded that the readings are false. I could not use the power calibration commands from tasmota.

Now I use template without power meter from
[https://templates.blakadder.com/zuczug_1gang.html](url)

Shaman2, 2021-02-16:
Hi. Somebody write this 2-gang W6B switch to ESPHome? Can you show me code?

kwirk, 2021-02-16:
> Hi. Somebody write this 2-gang W6B switch to ESPHome? Can you show me code?

Not tested this on my 2-gang switch (as not currently installed), but this derived from my 1-gang one which is working well.

```yaml
status_led:
  pin:
    number: GPIO1

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO3
      inverted: True
      mode: INPUT_PULLUP
    id: button1
    on_press:
      then:
        - light.toggle: light1
  - platform: gpio
    pin:
      number: GPIO4
      inverted: True
      mode: INPUT_PULLUP
    id: button2
    on_press:
      then:
        - light.toggle: light2

light:
  - platform: binary
    name: "Light 1"
    id: light1
    output: relay1
  - platform: binary
    name: "Light 2"
    id: light2
    output: relay2

output:
  - platform: gpio
    id: relay1
    pin: GPIO13
  - platform: gpio
    id: relay2
    pin: GPIO14
```

Shaman2, 2021-02-17:
> 
> 
> > Hi. Somebody write this 2-gang W6B switch to ESPHome? Can you show me code?
> 
But radio rf433 channel after reflashing is not working?

Shaman2, 2021-02-18:
> ```yaml
> status_led:
>   pin:
>     number: GPIO1

Which header need?

esphome:
  name: parents_room_switch
  platform: ESP8265
  board: esp_wroom_02

right?  Or ESP8566? Chip on my board is ESP8565

kwirk, 2021-02-18:
> But radio rf433 channel after reflashing is not working?

Sorry, my model doesn't have rf433, so not sure. Possibly try extra GPIOs as binary sensors if not working.

> Which header need?
```yaml
esphome:
  name: parents_room_switch
  platform: ESP8266
  board: esp8285
```

Hellcube, 2021-03-11:
I have MOES  Wi-Fi + RF 1GANG switch (single live wire - no neutral) model: WS-EU-RF. I tryed every possible combination and i can't flash this device. I have switched RX-TX, tryed flashing with tasmotizer, pyflash. Tryed every combination with RST+ GND + GPIO0 to get it into flash mode with no luck. Tryed Tuya-convert. I'm out of ideas...

[image]
[image]
[image]

9magic, 2021-03-13:
Hi,

Check the thread for post
[image: @honzapatCZ] <https://github.com/honzapatCZ> honzapatCZ
<https://github.com/honzapatCZ> commented on Dec 21, 2020
<https://github.com/ct-Open-Source/tuya-convert/issues/721#issuecomment-749051313>

Shaman2, 2021-03-16:
> I have MOES Wi-Fi + RF 1GANG switch (single live wire - no neutral) model: WS-EU-RF. I tryed every possible combination and i can't flash this device. I have switched RX-TX, tryed flashing with [...]

Are you have solved the problem?

Hellcube, 2021-03-16:
> Are you have solved the problem?

nope still no luck. I will try again today...

duvalt, 2021-03-16:
Hi Guys, I think the problem is linked to the USB / TTL converter you're using.
I never managed to flash WIFI-W6B board with FT232RL and it works fine with CP2102.
Hope it will help you

Shaman2, 2021-03-16:
> Lately I am using [this thing](https://www.aliexpress.com/item/32923592694.html?spm=a2g0o.productlist.0.0.702e7169CwKQoc&algo_pvid=null&algo_expid=null&btsid=0b0a555b16158952512111996e6cab&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) as my default flash device, works almost every time for everything.

CH340G no need to change driver for work?

lubino123, 2021-03-17:
What About 433? Do you know pins?  :/

EDV64, 2021-03-18:
Hi Guys,
I also had a long time to try flash this WiFI-W6B. My solution is to unsolder and lift up pin 11 of unmarked chip (see yellow route on screen above). This route is connected to U0RXD (pin 25) of ESP8285 and probably blocking UART exchange.
Also don't forget about GPIO0.
Good luck all.

xkolm036, 2021-03-25:
Thanks @EDV64 it works for me. I was not able to flash WIFI-W6B but after unsolder  pin 11 tasmotizer flash new bin file without any problem.

Shaman2, 2021-03-26:
> 
> 
> Thanks @EDV64 it works for me. I was not able to flash WIFI-W6B but after unsolder pin 11 tasmotizer flash new bin file without any problem.

pin 11 need solder after flash or leave unsolder? What pin 11 do?

xkolm036, 2021-03-26:
> 
> 
> > Thanks @EDV64 it works for me. I was not able to flash WIFI-W6B but after unsolder pin 11 tasmotizer flash new bin file without any problem.
> 
> pin 11 need solder after flash or leave unsolder? What pin 11 do?

My scenario was: 
1. Unsolder pin 11 on unmarked chip (marked by orange line on the picture)
2. Connect pins 3v3, RX, TX, GPIO0 (connect to the ground) and GND to USB-UART adapter (in my case NodeMcu v2 CP2102)
3. Connect USB-UART adapter to PCs USB
4. Flesh new bin file by Tasmotizer
5. Solder back pin 11.

And I have no idea what is the funtion of pin 11 I just try it.

EDV64, 2021-03-27:
> > Thanks @EDV64 it works for me. I was not able to flash WIFI-W6B but after unsolder pin 11 tasmotizer flash new bin file without any problem.
> 
> pin 11 need solder after flash or leave unsolder? What pin 11 do?

The reason of action is so the pin 11 connected to pin 25 of ESP8266, which is U0RXD of UART. If output level signal of pin 11 unmarked chip will low, RX line of ESP will not receive data from UART.
But don't forget about 25 pin of ESP8266 is also GPIO3, so it was possible a normal sensor or data exchange between unmarked chip and ESP on this line. Probably it can be a output of touch sensor in 3-way switch... don't know, I have 1-way one.
Resuming - if you plan regulary flashing throw UART, you don't have to solder pin 11 back. And alternatively, if all future updates will be throw OTA, then you can return this pin back.

Hellcube, 2021-03-28:
@EDV64 worked for me to Thank You! all works fine even 433 control. I will hook it up to the wall tomorrow.
[image]
[image]
[image]
[image]
I made a backup if some one needs it.
[Moes 1gang WiFI-W6B backup.zip](https://github.com/ct-Open-Source/tuya-convert/files/6217188/Moes.1gang.WiFI-W6B.backup.zip)

Thanks again every one.

Shaman2, 2021-03-28:
> worked for me to Thank You! all works fine even 433 control. I will hook it up to the wall tomorrow.

May better only cut off this line for you? Are you found what for this 11 contact?

Somebody try to use buzzer for tasmota or esphome beep function?

mincaeuro, 2021-06-07:
Not sure if some tried this, but I was able to flash it without touching the chip:
- wired as on pics UP (vcc->3.3v, rx->tx,tx->rx, gnd->GPi0)
- Hit Tasmotize Button (it started to "ping" the serial port or something like that...not sure)
- and shorted GND and RST on board
- and LED started to blink
- flashed successfully

Shaman2, 2021-07-12:
>     * (vcc->3.3v, rx->tx,tx->rx, gnd->GPi0)

gnd from UART adapter to gpio0? You are not connected gnd on W6B board?

EDV64, 2021-09-09:
Update for my previous posts - today I received a slightly different version of WiFI-W6B:

[image]

In this PCB we can see a different orientation of unmarked chip. After tracing I make sure that pin 5 (not 11 as earlier!) is connected to pin 25 (U0RXD) of ESP8285. When I applied the previous method and lift up pin 5 of unmarked chip, flashing was started.

Wain-PC, 2021-11-03:
Hi there. I've recently received a batch of Esooli / Minitiger branded switches with 1, 2 and 3 gangs.
All of them have the newer board version which is pictured by @EDV64 in the previous post. 
However, I was able to flash them with Tasmota using esptool.py without lifting any pins.

Procedure:
1. Permanently connect GPIO0 pin to GND.
2. Connect 3.3V, GND, RX, TX to UART.
3. Run esptool.py
4. Right after the tool has started briefly connect RST pin with GND. I was using a small screwdriver as the pins on the board are next to each other. The board was recognized by esptool only after this step.

Here are my Tasmota templates for every switch:

```
{"NAME":"1 Gang","GPIO":[0,288,0,0,0,32,0,0,224,0,0,0,0,0],"FLAG":0,"BASE":1}
{"NAME":"2 Gang","GPIO":[0,288,0,32,33,0,0,0,0,224,225,0,0,0],"FLAG":0,"BASE":1}
{"NAME":"3 Gang","GPIO":[0,52,0,17,19,18,0,0,22,21,23,0,0],"FLAG":0,"BASE":1}
```

leranp, 2021-11-21:
The RST was working for me, without pulling up the PIN in the chip,
But i have a random ON/OFF in the switches , i even tried to connect only the PCB and connect to 3.3V , does someone have the same problem? 
Tried to remove the RF Receiver , but the problem isn't gone

EDV64, 2022-02-28:
> Please tell me, is this the radio frequency part of 433MHz? [image]

"Highly-likely" yes, because the RF (WiFi) antenna interface of ESP8285 has pin2 and connected throw smd to other PCB structure (not shown on your photo, but you can see it on my previous photo in the left-upper corner)

jernest0, 2022-05-30:
Hello. I recently received from the EARU brand with 1 Switch
The newest board version shown in @EDV64.

I was able to flash them with Tasmota using esptool.py/Tasmnotizer without desoldering any pins.

Process:
As indicated by @Wain-PC

The RF system is not affected by the Firmware change, I have been able to add and remove RF remotes after flashing them with the Manufacturer's instructions.
This is the template that worked for me

{"NAME":"WIFI-W6B-1","GPIO":[0,544,0,0,0,32,1,1,224,0,0,0,0,0],"FLAG":0," BASE":18}

Elkropac, 2023-01-17:
> Update for my previous posts - today I received a slightly different version of WiFI-W6B:
>
> [image]
>
> In this PCB we can see a different orientation of unmarked chip. After tracing I make sure that [...]

Hi, i think, i have this version of PCB. Have you managed to get it flashing without unsoldering pin 5 from unmarked chip?. I tried shortening RST pad (the one between GPI0 and GND) to ground but no luck

Elkropac, 2023-01-18:
Hi, i tried pin 25 and it was connected like on your picture, I did not try pin 26.
I also think, that the pad marked as RST on pictures is connected to pin near on corner with dot of unlabeled chip, but shortening it to ground did not make any difference

EDV64, 2023-01-18:
> Hi, i tried pin 25 and it was connected like on your picture, I did not try pin 26. I also think, that the pad marked as RST on pictures is connected to pin near on corner with dot of unlabeled chip, but shortening it to ground did not make any difference

It was great if there was two SEPARATE RST pins - one in ESP and other in unlabeled chip. So you can connect second pin to ground and noname chip will be disabled...

Elkropac, 2023-01-18:
> It was great if there was two SEPARATE RST pins - one in ESP and other in unlabeled chip. So you can connect second pin to ground and noname chip will be disabled...

Just checked it and RST connects to pin 32 on ESP too, so it cannot be used , i guess? So i need to unsolder the pin 5 on unlabeled chip ... oh boy

pritam2016bakshi, 2023-03-07:
Do you have 4 gang switch firmware?

[42 more replies]