ct-Open-Source/tuya-convert #722: LSC Bulb e14

cognacais, 2020-07-30:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-07-30:
Please follow the instructions and post your logs, without doing this I cannot help

cognacais, 2020-07-31:
Hi

Sorry, it's not a good Idea to publish issue between 2 professionnal meetings. 😉

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5006475/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5006476/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5006477/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5006478/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5006479/smarthack-mqtt.log)

Thanks you

kbfifi, 2020-08-02:
Hi cognacais, did you buy the LSC E14 color ones? They seem to be Realtek not ESP. kueblec can you confirm?

kbfifi, 2020-08-02:
Just found out that it should work:
https://templates.blakadder.com/lsc_smart_connect_C45.html

Anyway I get the same message as cognacais: "Device did not appear with the intermediate firmware"

cognacais, 2020-08-02:
Hi kbfifi

Yes, it's this model. And i flashed 2 bulbs last month and integrated to home assistant.

kbfifi, 2020-08-02:
Meanwhile I can confirm it's an esp by a simple network scan on the vtrust-flash network. Probably there's a firmware update that needs to have a new solution. @kueblc I read something about a product key?

kbfifi, 2020-08-02:
To be complete my files as well:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5012744/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5012745/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5012746/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5012747/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5012748/smarthack-wifi.log)

kueblc, 2020-08-02:
> Just found out that it should work:
> https://templates.blakadder.com/lsc_smart_connect_C45.html
> 
> Anyway I get the same message as cognacais: "Device did not appear with the intermediate firmware"

This doesn't mean it should work, as manufacturers have been using the same model numbers and FCC ID even if they change the chipset. It is quite possible that a device on the template repository has been switched to a Realtek or another chip.

> Probably there's a firmware update that needs to have a new solution. @kueblc I read something about a product key?

You're probably talking about #483, where the PSK format changed, which is not the same as a product key. The product key simply identifies the product, it is not a key in the cryptographic sense. All devices of the same make and model share the same product key.

For both of you @cognacais and @kbfifi, it looks like the process stopped at the token registration stage, which likely indicates the plaintext response is being rejected. This is probably a new patch by Tuya, so there's unfortunately not much we can do at this time. Hopefully we can develop a workaround.

kbfifi, 2020-08-02:
Thanks for your comments. Is there something I can do to speed this up?

cognacais, 2020-08-03:
Thanks you @kueblc for your job. There is any Hope ?

I have the same problem with this model https://templates.blakadder.com/calex-429004.html

I don't have log here. I Can post logs in 2 weeks (vacation 😉)

Same that @kbfifi if i Can help.

SquadScope, 2020-08-05:
Hey Guys,
may I can give some additional information. I bought some LSC RGBW E14 bulbs and successfully flashed them with tasmota some weeks ago. Then the bulbs were sold out for weeks in kinda all markets on the Niederrhein. Now I bought 2 additional bulbs and flashing is not possible anymore.
Maybe someone should remove them from compatible devices list?

I don't believe they changed chip to Realtek since macvendors.com says it's still Espressif Inc.

kbfifi, 2020-08-05:
Unless they are not or  no Ionger flashable by the native app, this problem is solvable I think. @kueblc Let me know if I should make tcpdump or so

kbfifi, 2020-08-07:
Update: Now I also bought the LSC E14 smart filament LED. I successfully flashed Tasmota with tuya-convert. However Tasmota does not connect to my network

cognacais, 2020-08-12:
Hey guys,

i have an information. The model 899807 (lsc e14) doesn't change but version has changed (V2 to V2.3)
https://templates.blakadder.com/lsc_smart_connect_C45.html

I found lsc e14 V1 do you know if it's possible to flash tasmota?

Thanks you

nbeernink, 2020-08-16:
I can confirm that the LSC C45 400lm RGBW Bulb v2.3 (look underneath the box) did not want to convert. My logs also show similar messages as to what @cognacais already posted.

I was able to convert the v2.0 version just fine.

cognacais, 2020-08-17:
Hi guys

Just for information bulbs v1 are flashable.

Have a good day

ameinert, 2020-08-25:
Dear Team and all,

i have the same issue with a newly bought LSC Smart Light Strip (Action). 
I can see in the logs that my mobile and a second device (most likely the strip, mac 68:57:2d:a2...) 
connect to the AP of my Raspi, but then the script stucks in repeating

_"Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................."_

and finally:
_"Device did not appear with the intermediate firmware"_

I tried several times in slow blinking  and fast blinking mode (3 times powercycle)
In fast blinking mode the blinking stops while the script runs, which indicates that
something happens. The device was never connected to the app or another network before.

Is it confirmed that the devices are delivered with a new stock firmware that 
prevents tuya-convert to work?

Thanks and best regards!

nbeernink, 2020-08-27:
This bug report is not about the light strip but about the LSC e14 color bulb. 

Have a look here:
https://github.com/ct-Open-Source/tuya-convert/issues?q=is%3Aissue+lsc+strip+is%3Aclosed+

henkiejan1, 2020-09-15:
There is also a new PIR sensor what indicates exactly the same problem like above. It´s a new 360degrees sensor. It´s technical name is WiFi-702 PIR. Also sold by Action with the LSC brand. See http://go.quwl.net/h-pd-1221.html for a chinese product page.

solidminds, 2020-11-23:
Any updates on this problem? I also bought soms v2.3 bulbs and I hope there is a solution for it already...