ct-Open-Source/tuya-convert #727: Aoycocr U3S plugs not flashing

fysmd, 2020-08-02:
I bought 2 four packs of these from Amazon last week (UK).
Six of the eight devices have worked perfectly, two will not respond to their button when powered on, can't get into fast flash mode.

I have tried rerunning install_prereq.sh and running again (on a pi4)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5012427/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5012428/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5012429/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5012430/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5012431/smarthack-wifi.log)

Shell output below but I notice at retry five, my other machine can no longer ping the pi.

```
pi@raspberrypi:~/tuya-convert $ ./install_prereq.sh
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.6).
curl is already the newest version (7.64.0-4+deb10u1).
git is already the newest version (1:2.20.1-2+deb10u3).
haveged is already the newest version (1.9.1-7).
hostapd is already the newest version (2:2.7+git20190128+0c1e29f-6+deb10u2).
iw is already the newest version (5.0.1-1).
mosquitto is already the newest version (1.5.7-1+deb10u1).
net-tools is already the newest version (1.60+git20180626.aebd88e-1).
python3-dev is already the newest version (3.7.3-1).
python3-setuptools is already the newest version (40.8.0-1).
python3-wheel is already the newest version (0.32.3-2).
screen is already the newest version (4.6.2-3).
dnsmasq is already the newest version (2.80-1+rpt1).
libssl-dev is already the newest version (1.1.1d-0+deb10u3+rpt1).
python3-pip is already the newest version (18.1-5+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 90 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/drbild/sslpsk.git
  Cloning https://github.com/drbild/sslpsk.git to /tmp/pip-req-build-datrju8g
Requirement already up-to-date: paho-mqtt in /usr/local/lib/python3.7/dist-packages (1.5.0)
Requirement already up-to-date: tornado in /usr/local/lib/python3.7/dist-packages (6.0.4)
Requirement already up-to-date: pycryptodomex in /usr/local/lib/python3.7/dist-packages (3.9.8)
Building wheels for collected packages: sslpsk
  Running setup.py bdist_wheel for sslpsk ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-nhwdtcyy/wheels/19/5d/ca/bb92c9b2e4aa13e12306893a6eadd16dbfcdfca942476ecfa1
Successfully built sslpsk
Installing collected packages: sslpsk
  Found existing installation: sslpsk 1.0.0
    Uninstalling sslpsk-1.0.0:
      Successfully uninstalled sslpsk-1.0.0
Successfully installed sslpsk-1.0.0
Ready to start upgrade
pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
pi@raspberrypi:~/tuya-convert $`

```

Logs uploaded.

kueblc, 2020-08-02:
> Six of the eight devices have worked perfectly, two will not respond to their button when powered on, can't get into fast flash mode

It sounds like two of the devices are still running the intermediate firmware. Try running `start_flash` again, but this time do not connect a phone. Simply plug the device in and press enter.

fysmd, 2020-08-13:
Hi,
I have tried without connecting a phone and neither plug connects.  It just outputs same as above to the console 👍 Any other tips or might I have two dead devices?

duguk, 2020-08-15:
I don't think you have dead devices. I've bought the separate U3S plugs and had no problem. I've just bought a pack of the four and had the same problem that you have. I did compare them, and the font on the back of the plug is slightly bigger.

My log files are basically identical to yours, but I can upload them if it would help. I can also flash them directly/dump them if anything like that would help.

fysmd, 2020-08-16:
Were you able to get them back online?

duguk, 2020-08-16:
Yes, I think so. Mine came back into flash mode after power cycling and holding down the button. 
I haven't connected them to the official app (for fear of updates), but other than that they appear to be working normally.

All four of mine are the same, I have one apart and it looks identical inside; still the same ESP8266EX. I can dump the data but don't know how much help that would be.

lukejagger, 2020-10-31:
I have the same issue with some Aoycocr U2S smart plugs ordered from Amazon UK (Oct 2020).  Is it actually possible to flash these by connecting pins to the board itself if this WIFI method isn't working any more?

duguk, 2020-10-31:
> I have the same issue with some Aoycocr U2S smart plugs ordered from Amazon UK (Oct 2020). Is it actually possible to flash these by connecting pins to the board itself if this WIFI method isn't working any more?

It is possible, but it's difficult to access the pins and they're not labelled. These pins are on the reverse of this board (under the board between the live and earth terminals). The ESP board is mounted perpendicular into the mainboard and can be accessed underneath. A gentle squeeze in a vice will release the glue, but the rest of the board needs to be desoldered from the mains power to get to these. I was able to program one myself (I have a few!) but wasn't able to get it back together safely.

lukejagger, 2020-10-31:
Thanks @duguk. Sounds too tricky!  I have tried the alternative Tuya Convert fork as well, but that didn't work with the newer U2S either.

drumstick77, 2020-12-01:
hi All,

I'm also havin the same issue.
I bought U3S recently via amazon, none of them worked. so I returned it.
today received Avatar plugs and same issue is happening.
is there a limit now on the new version of these plugs?

[image]