ct-Open-Source/tuya-convert #728: Unable to flash Arlec ABL015HA 10W Bunker Light - device doesn't appear to connect to vtrust-flash

DarkStorm99, 2020-08-03:
Hi everyone,

hoping you can help out. I am running a raspberry pi 0 W on buster lite. I have tried quite a few things but the behaviour seems the same. Unfortunately I don't have anything else that supports AP wifi, as I get an error message stating that when I try live booting ubuntu on my laptop. I have also tried rasbian buster full and jessie full with the same results. I suspect its the lightas my current setup has flashed a few tuya based items previously. 

This is the device in question:
https://templates.blakadder.com/arlec_ABL015HA.html

I put device into EZ mode by turning on and off 3 times. Flashes rapidly. I then have tried with a few different phones, as well as 2 computers, to connect to vtrust-flash. No problems. The device just keeps flashing. Occasionally, the flashing stops, but there is no difference in the tuya-convert script, it just says retrying. 

Fresh logs attached from a new install of tuya-convert. 

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5015583/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5015584/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5015585/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5015586/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5015587/smarthack-udp.log)

I also saw the PSK thread but I don't have the same error. I do have this line though in the psk log:
could not establish sslpsk socket: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1056)

I saw this in the wifi log:
wlan0: Could not connect to kernel driver

Extensive searching of this github didn't give me any clues either. Spent a few hours trying different things. Any advice?
Unfortunately GPIO 0 is blocked by a cap, so a serial flash is borderline impossible. 
Thanks!

forstermatth, 2020-08-08:
I'm getting similar (the same) logs for the Globe Brand bulbs that I have previously successfully flashed. Looking at the wifi log, it looks like two devices are connecting (the proxy and the bulb), so perhaps there was an update? The bulb stops flashing, which I would take as it connecting, but the script doesn't manage to find it.

forstermatth, 2020-08-09:
https://github.com/ct-Open-Source/tuya-convert/issues/483

freedoh, 2020-10-05:
I can confirm that the Arlec ABL015HA uses the common TYWE2S ESP8266 module, and that the "10W" LED light is made up of 30x 3528 LEDs driven by a RM9010B LED driver chip that's controlled by PWM on GPIO14.

Original firmware attached.
[Arlec_ABL015HA_firmware.zip](https://github.com/ct-Open-Source/tuya-convert/files/5324901/Arlec_ABL015HA_firmware.zip)