ct-Open-Source/tuya-convert #729: LSC power plug "Device did not appear with the intermediate firmware"

mroelandts, 2020-08-04:
**name, model** = LSC smart power plug
This device is supported according to the [wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTPS-firmware)) and also according through [tasmota](https://templates.blakadder.com/lsc_smart_connect_power_plug.html).

I do not know how to retrieve the firmware version.
I have not yet connected it to the official app in order to not update the firmware.
(I though this would be beter as I want to flash tasmota on it.)
I'll connect it if its necessary.

Output:
```
tuya-convert v2.4.4
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...

Closing AP
Exiting...
````
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5022129/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5022130/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5022131/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5022132/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5022133/smarthack-mqtt.log)

kueblc, 2020-08-04:
Unfortunately, while this device may have been previously supported, the model you have is not.

According to the MAC address in your wifi log, the device does not use an ESP. This means it cannot run ESP firmware such as Tasmota and is not compatible with tuya-convert.

We've been seeing this a lot lately; manufacturers are switching chipsets without changing model numbers or FCC IDs.

closed 2020-08-04 by kueblc

mroelandts, 2020-08-04:
Hello, how can you find the chipset through the MAC address?
It is not connected to [issue 483](https://github.com/ct-Open-Source/tuya-convert/issues/483)?
I really was hoping to get this plug out of the cloud...

How can I buy a new plug that I can flash if you can't trust the manufacturers to not switch chipsets.
It says on the box that it's rev 2.0 and I read on a forum that it was possible to flash this plug.

kueblc, 2020-08-04:
> how can you find the chipset through the MAC address?

You can use an OUI lookup tool, [such as this one from Wireshark](https://www.wireshark.org/tools/oui-lookup.html)

> It is not connected to [issue 483](https://github.com/ct-Open-Source/tuya-convert/issues/483)?

No

> I really was hoping to get this plug out of the cloud...
> 
> How can I buy a new plug that I can flash if you can't trust the manufacturers to not switch chipsets.

I understand, as we all do. It is sadly not a priority for these companies but hopefully we can convince them otherwise. We need to be writing these companies and letting them know that there is a huge market for off cloud and open source IoT.

mroelandts, 2020-08-06:
I have added a note to the wiki, so that others do not have the same issue.
I'll also try to do this on the supported devices of the tasmota wiki

stefan123t, 2021-04-30:
@mroelandts I got the 970761 from Action in Germany bought today. It has two TX8 screws to easily pry it open inside is some WB2S Module custom built from Tuya. This is using a so called BK7321T chipset for which they offer their SDK here: https://github.com/tuya/tuya-iotos-embeded-sdk-wifi-ble-bk7231t

@kueblc maybe that also contains some detail information about the dreaded pre shared key everyone lacks for the "older" ESP8266/8265 based modules ?

stefan123t, 2021-04-30:
Especially two things caught my eyes:

This looks a bit like the prefix of the pskkey you were referring to in #483
https://github.com/ghsecuritylab/tysdk_for_bk7231t/blob/1cecb9af3b22dcc063311a55cfb7a5913799a284/project/tuya_iot/include/tuya_iot_sdk/tuya_cloud/gw_intf.h#L166
    BYTE_T dynamic_tls_mode;/*0:invalid 1:disable 2:psk_only 3:ecc_psk 4:ecc_only  */

and here are two prototype methods that might be interesting too:
https://github.com/ghsecuritylab/tysdk_for_bk7231t/blob/1cecb9af3b22dcc063311a55cfb7a5913799a284/project/tuya_iot/include/tuya_iot_sdk/tuya_cloud/iot_httpc.h#L22
OPERATE_RET httpc_gw_opeinfo_common_post_no_remalloc(IN CONST CHAR_T *url, IN CONST CHAR_T *api_name, IN CONST CHAR_T *api_ver, IN CONST CHAR_T *uuid, IN CONST CHAR_T *aes_key, IN CONST CHAR_T *gwid, IN CHAR_T *post_data, IN UINT_T data_total_len, IN CONST CHAR_T *p_head_other, OUT ty_cJSON **pp_result);

https://github.com/ghsecuritylab/tysdk_for_bk7231t/blob/1cecb9af3b22dcc063311a55cfb7a5913799a284/project/tuya_iot/include/tuya_iot_sdk/tuya_cloud/iot_httpc.h#L74
OPERATE_RET http_pskkey_get(OUT ty_cJSON **result);

stefan123t, 2021-04-30:
Sorry I searched the wrong github repo, but the tuya_iot sdk within the ghsecuritylab repo seems to be quite similar to the original from tuya themselves:
https://github.com/tuya/tuya-iotos-embeded-sdk-wifi-ble-bk7231t

Haraldvdh, 2023-06-17:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

stefan123t, 2023-08-03:
@Haraldvdh thanks for the tear-down pictures. They are probably appreciated by the folks at https://github.com/openshwprojects/OpenBK7231T_App/issues/172 and if you add it to their list of Beken BK7231T teardowns in the (mostly polish) electroda forum you may as well get an SD-Card.
Here is another (dutch) forum that focuses on this Smart Plug model: https://gathering.tweakers.net/forum/list_message/63909938#63909938