ct-Open-Source/tuya-convert #731: Can't flash LSC Smart Connect LED light

Xanderr, 2020-08-06:
Hi,

I recently bought an LSC LED light (RGB from Action) in the hope I could change its firmware using tuya-convert. Sadly when attempting to flash it, it gives me the following error: Device did not appear with the intermediate firmware.

Looking around in other issues, it seemed others also have seen this message. But based on their logs, it doesn't seem to be the same cause. These are mine: [logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/5036404/logs.zip).
(The logs may contains some AP setup errors due to NetworkManager not allowing me to use my adapter, but I fixed that).

From what I can see, the light connects to the created AP, performs one single request (/gw.json?a=s.gw.token.get) and than stops doing anything. This results in the error message above. After this request the light also stops flashing (not sure if that is normal). I can still ping the light however.

I tried to use tuya-convert in both Linux Mint and Ubuntu 18.04, with different wifi adapters, but nothing seems to work. Am I missing something, or is the light just not compatible somehow?

Xanderr, 2020-08-07:
I figured I might as well add my findings here. I took the bulb apart to see what kind of chip was in it. Here are the photos:

1. [image]

2. [image]

In the first photo you can see that it is an ESP8285 (may need to zoom in a little, it's really tiny). Behind the white PCB on photo 1 I see pins exposed for 3.3V, TX and RX however. This means I may be able to get a dump of the firmware (and flash the bulb manually). Also, is it just me or is the white PCD really sloppily attached?

nbeernink, 2020-08-16:
Which fitting does your lamp have? I see different tuya convert issues on the e14 and the e27 warm fitting. 

* LSC Smart Connect Color LED 400 lumen 5 watt e14 fitting v2.0 can be flashed just fine.
* LSC Smart Connect Color LED 400 lumen 5 watt e14 fitting v2.3 has issues described in the one you linked to: #722 
* LSC Smart Connect Color LED 806 lumen 9 watt e27 fitting v2.0 - has a realtek chipset and cannot be converted at all
* LSC Smart Connect White LED 1400 lumen 14 watt e27 fitting v1.0 - has PSK certificate issues as described in  #483

taste66, 2020-10-15:
Hi @Xanderr 
Did you have any success with the Idinio bulbs? They have some in our supermarket with production date 07/2020

Xanderr, 2020-10-18:
Hi @taste66,

I had some issues with the delivery of the bulbs (they sent me an empty box), and that took a while to get sorted - draining my motivation. However, today I actually tried flashing it with tuya-convert, and with success. It flashed without a problem and is now running the latest Tasmota.

The production date on mine is quite old actually, it's from 07/2019. Considering that Tuya is moving away from ESP based chips, I worry that the success rate of tuya-convert will drop significantly. So there's a good chance one from 07/2020 will not work, but you can always try one.

Personally I will be looking for another (cheap) approach to the LED lights, since I don't really like the Tuya lottery :p

taste66, 2020-11-01:
@Xanderr thank you for your response, sorry I somehow missed the notification so only saw it a moment ago. Good you where lucky, I am trying now with the Tuya smart app but also this does not find the bulb. It only works wit the LSC smart app.
My goaul is to integrate them with Home assistant as described here: https://github.com/rospogrigio/localtuya

boxemall, 2021-08-01:
> 
> 
> Which fitting does your lamp have? I see different tuya convert issues with different models.
> 
> My experiences:
> 
>     * LSC Smart Connect Color LED 400 lumen 5 watt e14 fitting v2.0 - can be flashed just fine.
> 
>     * LSC Smart Connect Color LED 400 lumen 5 watt e14 fitting v2.3 - has issues described in the one you linked to: [LSC Bulb e14 #722](https://github.com/ct-Open-Source/tuya-convert/issues/722)
> 
>     * LSC Smart Connect Color LED 806 lumen 9 watt e27 fitting v2.0 - has a realtek chipset and cannot be converted at all
> 
>     * LSC Smart Connect White LED 1400 lumen 14 watt e27 fitting v1.0 - has PSK certificate issues as described in  [New PSK format #483](https://github.com/ct-Open-Source/tuya-convert/issues/483)

kinda late to the party so i hope i still receive an answer here. i was lucky since my e14 seems to be of the v2.0 and i was able to flash tasmota and cofigure it to appear on my WiFi. i am also able to access the configuration. thing is i dunno how to go on from here due to not knowing which config to use. can you share yours maybe? so many endless GPIO configs and it's driving me crazy.... regards