ct-Open-Source/tuya-convert #735: Action LSC CCT GU10 support

deku-m, 2020-08-10:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
No as soon as i start the flash procedure it fails (blinking led to solid and keeps sending/retry packets in smartconfig mode)

**Do you have a question about configuring your newly flashed device?**
No, only how to get this supported/working.

**Did you have issues getting a device to flash?**
LSC GU10 CCT (WARM WHITE/WHITE LIGHT)

**Please remember to close your issue when the problem has been addressed!**

Additional context
https://www.action.com/globalassets/cmsarticleimages/76/02/3000267_8712879146223-111.png?preset=mediaSliderImageLargeHD

kueblc, 2020-08-10:
Please include your logs

deku-m, 2020-08-11:
I will look for the logs this weekend. 
They are created in script folder right?

deku-m, 2020-08-15:
Included logs

kueblc, 2020-08-15:
Unsupported firmware, see #483

closed 2020-08-15 by kueblc