ct-Open-Source/tuya-convert #739: I cannot flash bulb with firmware 1.0.0

elbueno222, 2020-08-12:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

I have flashed most of my tuya devices with tuya-convert including power strips, plugs and different bulbs, but I have one bulb that I cannot flash and I don't understand why because according to Tuya the version of the firmware is 1.0.0 (for both Main Module and MCU Module). As soon as the flash process starts the bulb stop blinking, but tuya-convert keeps executing the "Resending SmartConfig Packets" and finally it cannot find the intermediate firmware and shows this message:

```
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
```

This is the template for the bulb:
https://templates.blakadder.com/infray_9W.html

I have the same issue with other device (this blind controller: https://templates.blakadder.com/zemismart_YH002.html) but the firmware of that device according to Tuya is Main Module:3.1.4 and MCU Module:YH.1.5.0, so I guess that firmware cannot be flashed by tuya-convert yet, but I don't understand why this issue happen with a bulb with firmware 1.0.0

kueblc, 2020-08-12:
Firmware version doesn't really mean anything, but please do share your logs

elbueno222, 2020-08-12:
> Firmware version doesn't really mean anything, but please do share your logs

Really? I thought the firmware version was the key to know if a device could be flashed or not (like every time Tuya updated the firmware version then tuya-converter would need to be updated to catch up with the changes implemented by Tuya, so older firmwares would be more likely to accept the flash than new firmwares), isn't this true?

These are the logs:
[Raspberry.zip](https://github.com/ct-Open-Source/tuya-convert/files/5066060/Raspberry.zip)

kueblc, 2020-08-12:
> Really? I thought the firmware version was the key to know if a device could be flashed or not (like every time Tuya updated the firmware version then tuya-converter would need to be updated to [...]

Firmware version is not related to the Tuya SDK version. Different distributors use their own versioning systems.

> WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware

Your bulb does not use an ESP8266. You will not be able to flash this device with Tasmota.

> ID: 0242416f68626d64366147393149465231b8aa05cff42f1ffe375a12146cc1f945ea29984cffd41eae1db97cdc75a332dc

The blind controller is blocked by #483. This device can be flashed by serial.

closed 2020-08-12 by kueblc