ct-Open-Source/tuya-convert #741: Can't flash 1 of 3 plugs (Deltaco SH-P01E, similar to Gosund SP1 et al)

abstrakct, 2020-08-14:
Hello, I bought 3 Deltaco SH-P01E smart plugs, and have flashed two of them successfully using tuya-convert. But I'm unable to get the third one to work. When tuya-convert starts talking to the plug the blue led quickly stops flashing and tuya-convert keeps on "Resending SmartConfig Packets" until it gives up.

They were all bought together at the same store, so one would assume they're identical, but I see now that two of the boxes have "PO183437" printed on the bottom, but the third box has "PO184459" printed instead. I don't know which plug was in which box though. But maybe one plug is from a newer batch which can't be flashed?

For the record, I don't have the Deltaco app installed on my phone, so I didn't accidentally update the plug's official firmware or anything like that.

Logs:
[smarthack-psk.log](https://gist.github.com/abstrakct/4b3c5f7221eeb952d97c1602d004a699)

smarthack-udp.log:
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
```

smarthack-web.log:
```
Listening on 10.42.42.1:80
[I 200814 19:27:08 web:2250] 302 GET /generate_204 (10.42.42.11) 0.56ms
[I 200814 19:27:08 web:2250] 302 GET /generate_204 (10.42.42.11) 0.32ms
[I 200814 19:27:08 web:2250] 200 GET / (10.42.42.11) 0.51ms
```

smarthack-mqtt.log:
```
1597433210: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1597433210: Using default config.
1597433210: Opening ipv4 listen socket on port 1883.
1597433210: Opening ipv6 listen socket on port 1883.
^C1597433748: mosquitto version 1.4.15 terminating
```

[smarthack-wifi.log](https://gist.github.com/abstrakct/0b207450658573241157d0d814e08bb1)

For the plugs that I did successfully flash, here's the device-info from the backups tuya-convert made:
```
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: aa23d4
MAC: C4:4F:33:AA:23:D4
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
```
I can provide the logs from the successful flashes as well if that's useful.

Any idea as to where the problem is? Is the third plug "unflashable", or is it possible tuya-convert will support it in the future? Let me know if there's any more info or logs I can provide.

kueblc, 2020-08-14:
Unfortunately not supported at this time. See #483

If you still have the firmware backups made by tuya-convert for the other devices, please consider sharing these as this may help us as we search for a workaround. Also let us know if you are willing to open a device for serial flashing.

abstrakct, 2020-08-15:
OK, thank you for the information! I still have the firmware backups:

[device 1](https://filebin.net/6t3m7wkldqmvulsd/20200813_173046.tgz)
[device 2](https://filebin.net/6t3m7wkldqmvulsd/20200814_190503.tgz)

I might be able to try serial flashing, I have a usb thing with a CH340G chip. But there are no screws on the smart plug so I'll have to "destroy"

abstrakct, 2020-08-15:
OK, thank you for the information! I still have the firmware backups:

[device 1](https://filebin.net/6t3m7wkldqmvulsd/20200813_173046.tgz)
[device 2](https://filebin.net/6t3m7wkldqmvulsd/20200814_190503.tgz)

I might be able to try serial flashing, I have a usb thing with a CH340G chip. But there are no screws on the smart plug so I'll have to "destroy" it in order to open it up. I'll let you know if I do it.

(sorry for closing and reopening, apparently Alt+C does that?)

Mirodin, 2020-11-11:
This happend to me to. 3 out of my 4 SP1 plugs I bought of Amazon did go through flashing just fine. But the 4th does not work as it exhibits the same behaviour as OPs plugs. Unfortunately I am terrible at soldering (tried today for over 1 hour and just created cold joins that would fall apart by themselves. As I am afraid of killing my plug by fruitless trying I stopped my attempts.

becmar39, 2020-12-02:
Same for me. 2 out of 4 Gosund SP1 from Amazon can not get flashed according to #483

MrDemocracy, 2023-02-09:
> I might be able to try serial flashing, I have a usb thing with a CH340G chip. But there are no screws on the smart plug so I'll have to "destroy" it in order to open it up. I'll let you know if I do it.

The plug has 4 screws. Two under the lid between the socket prongs and two inside the plug on the copper brackets that you have to remove before you can unclip and lift out the motherboard.

pritam2016bakshi, 2023-03-02:
where your backup files? link expire.

pritam2016bakshi, 2023-03-02:
> OK, thank you for the information! I still have the firmware backups:
> 
> [device 1](https://filebin.net/6t3m7wkldqmvulsd/20200813_173046.tgz) [device 2](https://filebin.net/6t3m7wkldqmvulsd/20200814_190503.tgz)
> 
> I might be able to try serial flashing, I have a usb thing with a CH340G chip. But there are no screws on the smart plug so I'll have to "destroy" it in order to open it up. I'll let you know if I do it.
> 
> (sorry for closing and reopening, I accidentally hit a keycombo that saves your comment and closes the issue..)

where your backup files? link expire.