ct-Open-Source/tuya-convert #742: Device did not appear with the intermediate firmware

wernerhp, 2020-08-14:
I have 2 Smart Sockets that I'm trying to flash.
Qualitel Smart Wi-Fi Plug with power meter
Model IO-WIFI-PlugSA
Espressif Inc.

1. I connect my phone to `vtrust-flash`
2. I press and hold the power button to enter fast blink mode.
3. I hit ENTER.
After a couple of seconds the LED stops blinking.
The script just retries and fails.

```
$ docker-compose exec tuya start
tuya-convert v2.4.4
Checking for network interface wlp8s0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED
will blink fast). This is usually done by pressing and holding the
primary button of the device
   Make sure nothing else is plugged into your IoT device while
attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.......
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] N
======================================================
Cleaning up...
Closing AP
Exiting...
```

Logs
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5077255/smarthack-mqtt.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5077256/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5077257/smarthack-web.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5077258/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5077259/smarthack-wifi.log)

pmflav, 2020-08-15:
Having the same issue

MrLemur, 2020-08-15:
This sounds like #483

pmflav, 2020-08-15:
Same. But that issue was back in Jan.

pmflav, 2020-08-15:
> This sounds like #483

Same. But that issue was back in Jan.

MrLemur, 2020-08-15:
It's still not been fixed, and probably won't for a long while.

kueblc, 2020-08-15:
Yes it is indeed #483 and there is no certainty that it is possible to work around this.

closed 2020-08-15 by kueblc

chester512, 2020-08-31:
FYI - I have had the same issue with 4 out of 6 Novostella 20kw Flood Lights i purchased from amazon in the last 2 weeks.

Noschvie, 2020-08-31:
@chester512 : do you have a "how to" to open and flash the "Novostella 20kw Flood Light" in case of OTA isn't working ?