ct-Open-Source/tuya-convert #745: OSError: [Errno 99] Cannot assign requested address

membersound, 2020-08-18:
I'm connected to trust-flash with my phone, and everthing seems fine. Anyways I'm getting the following error. What could I do?

```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
Traceback (most recent call last):
  File "./smartconfig/main.py", line 26, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/usr/bin/tuya-convert/scripts/smartconfig/smartconfig.py", line 46, in smartconfig
    sock = SmartConfigSocket()
  File "/usr/bin/tuya-convert/scripts/smartconfig/smartconfig.py", line 29, in __init__
    self._socket.bind((address, 0))
OSError: [Errno 99] Cannot assign requested address
........................................................................
```

kueblc, 2020-08-18:
Can you access 10.42.42.1?

membersound, 2020-08-19:
Maybe it is due to using the internal wifi card. I switched to using a wifi stick, there the AP startup works.

closed 2020-08-19 by membersound

sanderlv, 2020-09-30:
I have a Pi4 and have the same with kali etc.

I ordered a pi4 due to the fact it did not work with a p2 and external SD card...

BUMMER.

```
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
Traceback (most recent call last):
  File "./smartconfig/main.py", line 26, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/home/kali/tuya-convert/scripts/smartconfig/smartconfig.py", line 46, in smartconfig
    sock = SmartConfigSocket()
  File "/home/kali/tuya-convert/scripts/smartconfig/smartconfig.py", line 29, in __init__
    self._socket.bind((address, 0))
OSError: [Errno 99] Cannot assign requested address
........................................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]
```

kueblc, 2020-09-30:
> OSError: [Errno 99] Cannot assign requested address

Make sure the address is available

sanderlv, 2020-09-30:
> 
> 
> > OSError: [Errno 99] Cannot assign requested address
> 
> Make sure the address is available

Thank you for your reply! what address? How and where? sorry, noob here

kueblc, 2020-09-30:
10.42.42.1. Try seeing if you can ping it or access it in a web browser. Make sure `hostapd` started and you can connect to the `vtrust-flash` AP with another device. Make sure that device gets an IP address assigned in the 10.42.42.x subnet.

kueblc, 2020-09-30:
Also see the resolution the author of this issue posted.

> Maybe it is due to using the internal wifi card. I switched to using a wifi stick, there the AP startup works.

sanderlv, 2020-09-30:
vtrust-flash I cannot see with any device. Not appearing. I have tried a pi2 with external dlink adapter, that was also no luck. I was told a pi4 would work with built in wifi...

kueblc, 2020-09-30:
Clearly this is the issue. Try another adapter or seek assistance from the raspberry pi community getting `hostapd` working.

sanderlv, 2020-09-30:
These scripts were tested in

     a Raspberry Pi 3B / 3B+ / 4B with Raspbian Stretch and its internal Wifi chip

sanderlv, 2020-09-30:
After 3rd attempts... installed raspberri pi os and now I am 1 step further. now starting an ap. cann sonnect to it via my phone and when going to 10.42.42.1 I see ÿou are connected to vtrust-flash". But nothing happens when I try to press the button on my esp devie for 5 secs... only getting:

```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N]

```

What next :-(

kueblc, 2020-09-30:
> Check the *.log files in the scripts folder

Please make future comments on your own issue to spare this issue author's email from blowing up.