ct-Open-Source/tuya-convert #747: How to flash Gosund SP111

membersound, 2020-08-19:
I'm unable to flash the SP111 device OTA.
The AP starts up, smartphone connects, sp111 is set into discovery mode (blue light blinks fast).

As soon as I start the flashing routine, the blinking stops, and the docker container only shows the following output. What could I do?

```

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..........
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] 

```

kueblc, 2020-08-19:
Looks like your device is running the intermediate firmware. Restart the process but this time do not connect a phone, just plug the device in and press enter.

membersound, 2020-08-19:
Thank you so much! That worked for both of my devices.

closed 2020-08-19 by membersound