ct-Open-Source/tuya-convert #748: Docker Build - "Not Found" Error in install_prereq.sh

jtrout, 2020-08-19:
I'm trying to run the process described under [Using Docker](https://github.com/ct-Open-Source/tuya-convert#using-docker) to get this up and running in Windows 10, but step 6 in the Docker Build is failing as follows:

```
PS C:\source\repos\tuya-convert> docker build -t tuya:latest .
Sending build context to Docker daemon  5.267MB
Step 1/7 : FROM phusion/baseimage:master
 ---> 4fc88116db8c
Step 2/7 : RUN apt-get update && apt-get install -y sudo iproute2 iputils-ping
 ---> Using cache
 ---> fc6c93bcb265
Step 3/7 : RUN echo '* libraries/restart-without-asking boolean true' | sudo debconf-set-selections
 ---> Using cache
 ---> 6deb8c7edb58
Step 4/7 : COPY docker/bin /usr/bin/
 ---> Using cache
 ---> 0cc862c95091
Step 5/7 : COPY . /usr/bin/tuya-convert
 ---> a16976fccd15
Step 6/7 : RUN cd /usr/bin/tuya-convert && /usr/bin/tuya-convert/install_prereq.sh
 ---> Running in a95eebd0102d
/bin/sh: 1: /usr/bin/tuya-convert/install_prereq.sh: not found
The command '/bin/sh -c cd /usr/bin/tuya-convert && /usr/bin/tuya-convert/install_prereq.sh' returned a non-zero code: 127
```

Any ideas what I might check? Is it possible a command in the install_prereq.sh itself is failing?

Thanks for any guidance!

EltrinEU, 2020-08-26:
In file install_prereq.sh change line endings from windws CR/LF to Linux. You can change it by using Notepad++ or other text editor hat allows for line ending change.

jtrout, 2020-08-27:
Perfect, that did the trick, thanks so much!

closed 2020-08-27 by jtrout