ct-Open-Source/tuya-convert #752: donor device keeps connecting and disconnecting from vtrust-flash

jjsander, 2020-08-24:
I'm sorry to take up space here, but I've combed through the Issues list and have not found a solution that works with my device. I'm trying to flash a Gosund (Smartlife?) WP3 (I do not know firmware) but I can't get to a stable WiFi connection between my donor devices (I've tried 2 - my Android phone and a Debian machine running jessie) and the flashing device (RasPi 3B+ with an ethernet connection and running the latest version of Octopi / Debian Buster). The donor device keeps connecting / disconnecting every couple of seconds. I can't tell if it's pulling an IP address from the Tuya device or not. It does not appear that my Tuya device is running firewalld or ufw. I've tried with and without any services that look like they'd cause DNS or DHCP lease issues. I may just be doing something really dumb...anyone with fresh eyes would be most welcome and appreciated. 

Logs below. 

Thanks in advance!

Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 19312.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
^C======================================================
Cleaning up...
Closing AP
Exiting...
`
smarthack-wifi.log:

System info
Linux octopi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:fb:9a:7e and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED <mac address>
wlan0: AP-STA-DISCONNECTED <mac address>
wlan0: AP-STA-CONNECTED <mac address>
wlan0: AP-STA-DISCONNECTED <mac address>
wlan0: AP-STA-CONNECTED <mac address>
[... 48 lines cut ...]
  rng-tools.service                                           loaded active running rng-tools.service
  rpi-eeprom-update.service                                   loaded active exited  Check for Raspberry Pi EEPROM updates
  rsyslog.service                                             loaded active running System Logging Service
  ssh.service                                                 loaded active running OpenBSD Secure Shell server
  systemd-fsck-root.service                                   loaded active exited  File System Check on Root Device
  systemd-fsck@dev-disk-by\x2dpartuuid-6c586e13\x2d01.service loaded active exited  File System Check on /dev/disk/by-partuuid/6c586e13-01
  systemd-journal-flush.service                               loaded active exited  Flush Journal to Persistent Storage
  systemd-journald.service                                    loaded active running Journal Service
  systemd-logind.service                                      loaded active running Login Service
  systemd-modules-load.service                                loaded active exited  Load Kernel Modules
  systemd-random-seed.service                                 loaded active exited  Load/Save Random Seed
  systemd-remount-fs.service                                  loaded active exited  Remount Root and Kernel File Systems
  systemd-sysctl.service                                      loaded active exited  Apply Kernel Variables
  systemd-sysusers.service                                    loaded active exited  Create System Users
  systemd-timesyncd.service                                   loaded active running Network Time Synchronization
  systemd-tmpfiles-setup-dev.service                          loaded active exited  Create Static Device Nodes in /dev
  systemd-tmpfiles-setup.service                              loaded active exited  Create Volatile Files and Directories
  systemd-udev-trigger.service                                loaded active exited  udev Coldplug all Devices
  systemd-udevd.service                                       loaded active running udev Kernel Device Manager
  systemd-update-utmp.service                                 loaded active exited  Update UTMP about System Boot/Shutdown
  systemd-user-sessions.service                               loaded active exited  Permit User Sessions
  triggerhappy.service                                        loaded active running triggerhappy global hotkey daemon
  user-runtime-dir@1000.service                               loaded active exited  User Runtime Directory /run/user/1000
  user@1000.service                                           loaded active running User Manager for UID 1000
● webcamd.service                                             loaded failed failed  the OctoPi webcam daemon with the user specified config

jjsander, 2020-08-24:
Update: In the brief window I have before the client (Android phone) is disconnected from the AP, an IP in the 10.42.42.x range is being assigned.

jjsander, 2020-08-24:
Sorry to spam my own topic here, but a deeper dive into hostapd with debug level 0 indicates there's something going on with RADIUS. Is that normal / desired behavior?

wlan0: STA <mac address> IEEE 802.11: associated
wlan0: AP-STA-CONNECTED <mac address>
wlan0: STA <mac address> RADIUS: starting accounting session 4339B01F3F8DA734
wlan0: STA <mac address> IEEE 802.11: disassociated

jjsander, 2020-08-24:
And, lastly, if I give my Android device a static IP address, it looks like it  successfully hangs onto the connection, so I'm assuming that points to dnsmasq? Obviously my tuya device will need that dhcp lease if it's going to flash correctly?

jjsander, 2020-08-26:
Ok, it's working now! Not sure exactly what did the trick, but I edited my hostapd.conf to put the interface location there as well as change the channel number (I had experimented with different ones based on my WAPs loading). I also commented out the setup_ap line 
`#printf "$(cat hostapd.conf)\ninterface=$WLAN" | sudo hostapd /dev/stdin`
 and ran hostapd directly 
`sudo hostapd ./hostapd.conf`

Not sure why that would have changed anything, but something definitely clicked and stabilized my donor device connection. My Tuya device flashed perfectly after that. Thanks!

closed 2020-08-26 by jjsander

kotzer3, 2021-01-15:
just to complete this, only issue, which helped me, was running
$sudo raspi-config
and set the correct WiFi country. Before setting the country, the wifi connection via mobile immediately disconnects after log in.