ct-Open-Source/tuya-convert #753: Can I flash Tuya's device that's based on BK7231D WiFi chip?

adeptps, 2020-08-25:
I bought the Tuya's device https://m.aliexpress.ru/item/33047521274.html.

Can it be flashed by using Tuya convert project?

xsasx, 2020-08-26:
Same here have the 2 gang dimmer and 2 gang switch attached the logs

https://www.moeshouse.com/collections/switch-module/products/2-gang-2-way-wifi-smart-light-switch-diy-breaker-module-smart-life-tuya-app-remote-control-working-with-alexa-echo-google-home

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5128884/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5128885/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5128886/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5128887/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5128888/smarthack-wifi.log)

kueblc, 2020-08-26:
No, as mentioned in the README, we only support devices based on the ESP82xx.

closed 2020-08-26 by kueblc