ct-Open-Source/tuya-convert #754: WIFI AP not working for flashing

Dandub, 2020-08-26:
Hi,

I need some help with the tuya convert as I don't seem to manage to get my pc to broadcast the AP WIFI.
See below and please let me know if I did anything wrong.
Many thanks,
Dan 

dan@Dans-NUC:~$ cd tuya-convert
dan@Dans-NUC:~/tuya-convert$ sudo ./install_prereq.sh
[sudo] password for dan: 
Hit:1 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:2 http://ie.archive.ubuntu.com/ubuntu focal InRelease                      
Hit:3 http://ie.archive.ubuntu.com/ubuntu focal-updates InRelease              
Hit:4 http://ie.archive.ubuntu.com/ubuntu focal-backports InRelease            
Hit:5 http://security.ubuntu.com/ubuntu focal-security InRelease               
Get:6 https://packages.grafana.com/enterprise/deb stable InRelease [12.1 kB]   
Err:6 https://packages.grafana.com/enterprise/deb stable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8C8C34C524098CB6
Reading package lists... Done
W: GPG error: https://packages.grafana.com/enterprise/deb stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8C8C34C524098CB6
E: The repository 'https://packages.grafana.com/enterprise/deb stable InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
dan@Dans-NUC:~/tuya-convert$ nano config.txt
dan@Dans-NUC:~/tuya-convert$ sudo ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlp58s0... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 706.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

kueblc, 2020-08-26:
Hi Dan, could you post your wifi log please?

Dandub, 2020-08-26:
> Hi Dan, could you post your wifi log please?

Hi Kueblc, thanks for trying to help. Please see below log

[    5.760187] iwlwifi 0000:3a:00.0: Applying debug destination EXTERNAL_DRAM
[    5.900395] iwlwifi 0000:3a:00.0: Applying debug destination EXTERNAL_DRAM
[    5.970183] iwlwifi 0000:3a:00.0: FW already configured (0) - re-configuring
[    8.343465] rfkill: input handler disabled
[    9.562519] wlp58s0: authenticate with ac:22:05:59:73:75
[    9.572204] wlp58s0: send auth to ac:22:05:59:73:75 (try 1/3)
[    9.577665] wlp58s0: authenticated
[    9.581847] wlp58s0: associate with ac:22:05:59:73:75 (try 1/3)
[    9.618789] wlp58s0: RX AssocResp from ac:22:05:59:73:75 (capab=0x511 status=0 aid=5)
[    9.621444] wlp58s0: associated
[    9.662031] IPv6: ADDRCONF(NETDEV_CHANGE): wlp58s0: link becomes ready
[   13.670811] logitech-hidpp-device 0003:046D:4023.0003: HID++ 2.0 device connected.
[   15.796736] logitech-hidpp-device 0003:046D:4054.0004: HID++ 4.5 device connected.
[   23.072091] Bluetooth: RFCOMM TTY layer initialized
[   23.072100] Bluetooth: RFCOMM socket layer initialized
[   23.072103] Bluetooth: RFCOMM ver 1.11
[   23.246895] rfkill: input handler enabled
[   25.910103] rfkill: input handler disabled
[   26.102323] audit: type=1400 audit(1598466738.726:53): apparmor="DENIED" operation="capable" profile="/snap/snapd/8790/usr/lib/snapd/snap-confine" pid=2100 comm="snap-confine" capability=4  capname="fsetid"
[   27.938124] audit: type=1400 audit(1598466740.562:54): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/mime/mime.cache" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   27.938127] audit: type=1400 audit(1598466740.562:55): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/mime/globs2" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   27.938129] audit: type=1400 audit(1598466740.562:56): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/mime/magic" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   27.938131] audit: type=1400 audit(1598466740.562:57): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/mime/aliases" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   27.938133] audit: type=1400 audit(1598466740.562:58): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/mime/subclasses" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   27.938134] audit: type=1400 audit(1598466740.562:59): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/mime/icons" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   27.938137] audit: type=1400 audit(1598466740.562:60): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/mime/generic-icons" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   28.410730] audit: type=1400 audit(1598466741.034:61): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/glib-2.0/schemas/gschemas.compiled" pid=2100 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   28.728939] audit: type=1107 audit(1598466741.350:62): pid=761 uid=103 auid=4294967295 ses=4294967295 msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.8" pid=2100 label="snap.snap-store.ubuntu-software" peer_pid=784 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[   29.770241] pci 0000:01:00.0: [8086:15da] type 01 class 0x060400
[   29.770304] pci 0000:01:00.0: enabling Extended Tags
[   29.770394] pci 0000:01:00.0: supports D1 D2
[   29.770395] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[   29.782149] pci 0000:02:00.0: [8086:15da] type 01 class 0x060400
[   29.782293] pci 0000:02:00.0: enabling Extended Tags
[   29.782451] pci 0000:02:00.0: supports D1 D2
[   29.782454] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[   29.782855] pci 0000:02:01.0: [8086:15da] type 01 class 0x060400
[   29.782992] pci 0000:02:01.0: enabling Extended Tags
[   29.783174] pci 0000:02:01.0: supports D1 D2
[   29.783180] pci 0000:02:01.0: PME# supported from D0 D1 D2 D3hot D3cold
[   29.783611] pci 0000:02:02.0: [8086:15da] type 01 class 0x060400
[   29.783736] pci 0000:02:02.0: enabling Extended Tags
[   29.783916] pci 0000:02:02.0: supports D1 D2
[   29.783921] pci 0000:02:02.0: PME# supported from D0 D1 D2 D3hot D3cold
[   29.784347] pci 0000:01:00.0: PCI bridge to [bus 02-39]
[   29.784367] pci 0000:01:00.0:   bridge window [mem 0xc4000000-0xda0fffff]
[   29.784380] pci 0000:01:00.0:   bridge window [mem 0xa0000000-0xc1ffffff 64bit pref]
[   29.784536] pci 0000:03:00.0: [8086:15d9] type 00 class 0x088000
[   29.784591] pci 0000:03:00.0: reg 0x10: [mem 0xda000000-0xda03ffff]
[... cut ...]
[  194.908929] iwlwifi 0000:3a:00.0: FW already configured (0) - re-configuring
[  198.859755] wlp58s0: authenticate with ac:22:05:59:73:75
[  198.867095] wlp58s0: send auth to ac:22:05:59:73:75 (try 1/3)
[  198.895917] wlp58s0: authenticated
[  198.899753] wlp58s0: associate with ac:22:05:59:73:75 (try 1/3)
[  198.937088] wlp58s0: RX AssocResp from ac:22:05:59:73:75 (capab=0x511 status=0 aid=5)
[  198.941186] wlp58s0: associated
[  198.995614] IPv6: ADDRCONF(NETDEV_CHANGE): wlp58s0: link becomes ready

kueblc, 2020-08-26:
There should be a `smarthack-wifi.log` file in your `tuya-convert/scripts` directory

Dandub, 2020-08-26:
> There should be a `smarthack-wifi.log` file in your `tuya-convert/scripts` directory

Sorry....

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
./setup_ap.sh: line 13: hostapd: command not found
Python 3.8.2
===========
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp58s0...
sudo: hostapd: command not found
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
System info
===========
94acc3b
Linux Dans-NUC 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
OpenSSL 1.1.1f  31 Mar 2020
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
./setup_ap.sh: line 13: hostapd: command not found
Python 3.8.2
===========
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp58s0...
sudo: hostapd: command not found
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...

Dandub, 2020-08-26:
found the problem,,,hostapd was missing now the AP is active so that's the good news
the bad news is that the BHT-002 thermostat doesn't find the AP od the AP doesn't find the thermostat...

any suggestions?

Thanks again for your help

Dandub, 2020-08-26:
A module seems to be missing according to the below....

Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/dan/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/dan/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
........................................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n

closed 2020-08-27 by Dandub