ct-Open-Source/tuya-convert #757: Did I brick my smart plug?

chrissv, 2020-08-30:
I have had much success running Tuya Convert on various Gosund smart plugs.

I started a session with a new plug (Gosund WP6, others of which I have successfully converted), it found the switch, performed a firmware backup, and it got to the menu where I could select return to stock, Tasmota, Espurna, or do nothing.

Instead of proceeding with the tasmota flash, I selected do nothing, expecting it to leave the switch back to the original (factory) state.

But now the plug appears unresponsive. Pressing the button doesn't toggle the relay, and pressing and holding doesn't cause the LED to flash at all. In fact, the only time the LED flashes is when I plug the plug in, it flashes momentarily.

If I run tuya-convert again, it appears to detect the switch, but it hangs at 0% trying to download the firmware, and times out. The program complains about not being able to reach the device thereafter.

Did I brick my socket?

chrissv, 2020-08-30:
I should have done some more searching. I found information in #709 which helped me get out of this intermediate state.

Maybe this information could be added to the "Troubleshooting" part of the Wiki...

closed 2020-08-30 by chrissv

kueblc, 2020-08-30:
Glad you recovered your device.

This is on the Troubleshooting page of the wiki, but feel free to make it more clear. The wiki is public.

kueblc, 2020-08-30:
> where I could select return to stock, Tasmota, Espurna, or do nothing

> I selected do nothing, expecting it to leave the switch back to the original (factory) state

Is there something unclear about the difference between "return to stock" and "do nothing"?

Let me know how this can be improved.

chrissv, 2020-08-30:
I think the part which was confusing is that there are 2 stages to the conversion - that if you "do nothing" the plug is left permanently in an in-between state (unless you either finish the process or do the /undo link).

To me "return to stock" means reflashing the unit with the downloaded backup firmware. That was the confusing part. But I understand now.

By the way, what is the process to un-tasmotize a plug and return it to the original Tuya firmware (which was backed up by this process)?